BillyBlaze/OctoPrint-TouchUI #74: TouchUI doesn't use OctoPrint's baseURL to generate paths

hugheaves, 2016-12-05:
It is possible to configure the OctoPrint server to run at a different URL using the mechanism's described here: https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-doesnt-work-when-connecting-via-a-different-url

For example, I run my OctoPrint server at "http://server_name/octoprint" because I run other services at the server root http://server_name/". This works fine for core OctoPrint and most plugins as it generates all the UI URL's using the "baseURL" setting.

However, the TouchUI plugin continues to generate URL's at the server root: "/plugin/touchui/static/ ... " instead of "/octoprint/plugin/touchui/static/ .."

**Reproduction Steps:**

1. Change baseURL to /octoprint (See : https://github.com/foosel/OctoPrint/wiki/Configuration#server )
2. Restart OctoPrint
3. Activate TouchUI interface

**Expected behavior:**
Touch UI functions properly.

**Observed behavior:**
Touch UI doesn't work at all. It is unable to load any of the CSS or other static resource files required for it to operate.

**Browser version:** Firefox 50, Chrome 54
**OctoPrint version:** 1.2.18 (master branch)
**TouchUI version:** 0.3.4

**Installed plugins:**
1. CuraEngine

BillyBlaze, 2016-12-05:
Thanks for reporting!

I just pushed a fix into the maintenance branch and will release this in 0.3.5 very soon.

closed 2016-12-30 by BillyBlaze

BillyBlaze, 2016-12-30:
It has fixed and released in 0.3.5

commit a552190e (2016-12-05): :art: Make PATHS relative - Fixes #74