BillyBlaze/OctoPrint-TouchUI #75: Potential webcam issues

amoose136, 2016-12-05:
TouchUI used to work with my webcam (the standard octoprint interface still does) but after some update the camera tab now is blank even though a timelapse is being made in the background. Has anyone else had this problem or am I the only one? My version of octopi (1.2.18) is running on an RPi 3 with TouchUI (0.3.4) and I typically access it via Chrome or Chromium.

BillyBlaze, 2016-12-15:
Hello, thanks for asking, sadly I am afraid you're the only one. I have upgraded everything to the latest versions and can still see the webcam feed.

amoose136, 2016-12-18:
I disabled a bunch of plugins and the webcam came back in touchui after restarting so it appears there is some incompatibility with my plugins. I'll try and isolate it.

amoose136, 2016-12-20:
So it appears that if the [Slicer plugin](http://plugins.octoprint.org/plugins/slicer/) is enabled, the webcam breaks only in ToughUI.

BillyBlaze, 2016-12-25:
Thanks for finding this out.

There are some known issues with Slicer and TouchUI running locally on a linux machine. Basicly Slicer does no feature detection and executes javascript which does not work. I expect eventually that this javascript API will work. But that is dependent on Chromium.

The quickfix would be to add feature detection, but that is not the task for TouchUI.

Salandora, 2017-02-22:
@BillyBlaze 
I have this Problem too hehe, but I tried to get a workaround/fix inside of TouchUI so I digged a bit around and found that this could really be a bug of touchui:

https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/master/octoprint_touchui/static/js/touchui.bundled.js#L1867

cloneTo: '.tab-content'
I think this should be changed, because .tab-content could be used of many plugins and therefore is to common to be reliable. 

I changed it to this localy:
cloneTo: '#tabs + .tab-content'
this makes sure that the wanted .tab-content element has a previous id='#tabs' sibling.

for me this solved the webcam not shown problem.

BillyBlaze, 2017-02-22:
Thanks for

BillyBlaze, 2017-02-22:
@Salandora, thanks for finding this out 👍 

I will fix this in the upcoming release!

BillyBlaze, 2017-04-16:
This has been fixed in the latest 0.3.7 release.

closed 2017-04-16 by BillyBlaze