BillyBlaze/OctoPrint-TouchUI #76: GUI no longer loading on HDMI touchscreen on Pi3 after system upgrades

NoNaym, 2016-12-14:
Versions: TouchUI 0.3.4 and OctoPrint 1.3.0, also tested in TouchUI 0.3.2 and OctoPrint 1.2.11

Problem: TouchUI GUI no longer loads on Raspberry Pi 3 with HDMI touchscreen after running "apt-get upgrade," instead I just boot up to the terminal prompting for login.

OctoPrint still loads up normally and is accessible from remote computers via any web browser, and when I run "service touchui status" it returns "active(exited)" which I want to say is normal, but I'm not 100% sure of that.

After messing with things for a while, my best guess is that one of the updates (maybe xserver related update) screwed something up... but I'm not quite sure how or where. For now I am reverting to my original working image, and refraining from running "apt-get upgrade," but I would think it would be best to be able to continue with the latest software updates... no?

BillyBlaze, 2016-12-14:
Thanks for reporting.

I just did an upgrade myself and I also encountered this issues.

I will investigate for a fix on a short notice.

NoNaym, 2016-12-14:
Thanks @BillyBlaze! Glad it was not just me

byronlcollier, 2016-12-14:
Aha, that would explain my problem. I've just set this up for the first time and I'm having the same issue: I'm presented with a console login and X/chromium don't autostart. Have run apt-get update as well as rpi-update, and I'm using the same TouchUI/OctoPrint versions as @NoNaym.

BillyBlaze, 2016-12-15:
I have found the culprit and fixed it in the autostart repository

To get everything working again after this upgrade please follow these steps:

1. Login with SSH and stop TouchUI

    ````
    sudo service touchui stop
    ````

2. Run

    ````
    cd ~/TouchUI-autostart/ && git pull
    ````

3. After you got the new files run:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo systemctl daemon-reload
    ````

4. Reboot or run ``sudo service touchui start``

Hope this works. I will also keep this thread open for anyone in the future that is going to upgrade ☹️

byronlcollier, 2016-12-15:
Confirmed working on a Pi2 using versions listed in OP. Pi boots directly to Chromium and loads TouchUI, as expected.

Many thanks for a swift fix.

NoNaym, 2016-12-19:
Seems to be working for me as well on my two Pi3s. Thanks again @BillyBlaze!!

closed 2016-12-30 by BillyBlaze

ruigmoreno, 2018-10-16:
### Configurations 

Raspberry Pi 3 Model B
5inch HDMI Display 800x480 pixel XPT2046 Touch Controller
Linux octopi 4.14.74-v7+ #1149 SMP Mon Oct 8 17:39:42 BST 2018 armv7l GNU/Linux

Total memory: 115
Used memory: 51
Free memory: 31
Total Swap: 99
Used Swap: 43

---
Hi all,

I did all steps that you told @BillyBlaze, but I dind't have successful. When I reboot system, It starts with the Raspbian's GUI figure, but the figure disappear and I access the same method by Putty using CLI.  I read a lot of forums. How can I solve this?

 [image]

BillyBlaze, 2018-10-16:
Disable the Raspbian GUI by either `sudo systemctl set-default multi-user.target` or by running `sudo raspi-config` and selecting 'Boot options' > 'boot to command'

Please create a new issue if you cannot resolve it.