BillyBlaze/OctoPrint-TouchUI #78: 90° rotated webcam image does not get positioned correctly

joris57, 2016-12-19:
After selecting the 'Rotate webcam 90 degrees counter clockwise' option, the webcam image does not show completely and partially overlaps the title bar, when the camera tab is selected. Strange is that when the browser is sent to the background (with the camera tab open) and then recalled, the display is ok.

BillyBlaze, 2016-12-25:
Thanks for reporting, I will look into this before the upcoming release.

BillyBlaze, 2016-12-29:
btw. do you have some more information about where this happened? (i.e. touchscreen, mobile, computer)
and which browser?

BillyBlaze, 2016-12-30:
if your quick I could fix this in the upcoming 0.3.5 👍

joris57, 2016-12-30:
This occurs on 
- any device (tablet, phone or PC)
- touch and non-touch
- Chrome, Microsoft Edge, default Samsung browser (whatever that is)
- cheap endoscope cam and Logitech C270

When switching between BROWSER tabs the image gets correctly positioned, when switching between TOUCHUI tabs (cam and any other tab), the image becomes messed up again.

BillyBlaze, 2016-12-30:
Thanks for the information, I was able to reproduce the problem and I will fix this in 0.3.5

BillyBlaze, 2016-12-30:
It has fixed and released in 0.3.5

closed 2016-12-30 by BillyBlaze