BillyBlaze/OctoPrint-TouchUI #79: xserver not working /running

Sodamaexchen, 2016-12-25:
Hi Bill,

i am trying to get the UI running as autostart, but im failing.
I followed yout guide to the letter (several times) but it gives me an error after executing `
sudo update-rc.d touchui defaults`

it says: "pi@octopi:~ $ sudo update-rc.d touchui defaults
insserv: Service xserver has to be enabled to start service touchui
insserv: exiting now!
update-rc.d: error: insserv rejected the script header
"

I used the newest available image of octopi, setup my touchscreen and installed the TouchUI plugin.
The TouchUI is working when i log in with my phone so the issue must be (probably) with the xserver.

I am not used to Linux and honestly stuck.
I do now a little programming like C or Basic but beeing an mechnical engineer i am out of my league.

Thanks in advance 
Max

BillyBlaze, 2016-12-25:
Hi Max,

Did you perhaps followed the wrong guide? It looks like it is missing a software package which is normally pre-installed on Jessie 'Full', however OctoPi is based on Jessie Lite.

If you did follow the right guide, then we might need to re-install the packages.

Sodamaexchen, 2016-12-25:
Thanks for the fast reply.

Happy Christmas by the way.

I used the guide here on github:
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

with the image of octopi from octoprint.org (octopi 0.13)

So i don't know where i took the wrong turn.

BillyBlaze, 2016-12-25:
Hmm okay, what is the ouput of this command?

``` 
dpkg-reconfigure -a 
```

Sodamaexchen, 2016-12-25:
pi@octopi:~ $ dpkg-reconfigure -a
Unknown option: a
Usage: dpkg-reconfigure [options] packages
  -u,  --unseen-only            Show only not yet seen questions.
       --default-priority       Use default priority instead of low.
       --force                  Force reconfiguration of broken packages.
       --no-reload              Do not reload templates. (Use with caution.)
  -f,  --frontend               Specify debconf frontend to use.
  -p,  --priority               Specify minimum priority question to show.
       --terse                  Enable terse mode.

BillyBlaze, 2016-12-25:
ugh, okay, next thing we can try. removing packages and re-installing them.

Run:
```
sudo dpkg -r xinit xserver-xorg xserver-xorg-video-fbdev x11-xserver-utils matchbox unclutter
sudo apt-get autoremove
```

After that, try:
```
sudo apt-get install --no-install-recommends xinit xserver-xorg xserver-xorg-video-fbdev x11-xserver-utils matchbox unclutter
```

Please pay attention to any errors during this process and please post them :)

Sodamaexchen, 2016-12-25:
ok, it did uninstall & install without any errors.

only one thing popped up, but i do not think its an issue:
"Setting up matchbox-window-manager (1.2-osso21-1) ...
update-alternatives: using /usr/bin/matchbox-window-manager to provide /usr/bin/x-window-manager (x-window-manager) in auto mode
Setting up libpango1.0-0:armhf (1.36.8-3+rpi1) ...
"

BillyBlaze, 2016-12-25:
ok that looks good, and now if you redo step 8 in my wiki, does that now work?

Sodamaexchen, 2016-12-25:
same error.....:
"
pi@octopi:~ $ sudo update-rc.d touchui defaults
insserv: Service xserver has to be enabled to start service touchui
insserv: exiting now!
update-rc.d: error: insserv rejected the script header"

i there something basics i am doing wrong ?

BillyBlaze, 2016-12-25:
No, I dont think so.

Let's try something else.

Run

```
sudo nano /etc/init.d/touchui
```

Then on linenumber 5 remove xserver

```
# Required-Start:     $all xserver
```

After that Save and Exit (CTR-O, Enter, CTR-X)

Then run:
```
sudo update-rc.d touchui defaults
```

Sodamaexchen, 2016-12-25:
different error:

pi@octopi:~ $ sudo update-rc.d touchui defaults
insserv: Script touchui is broken: incomplete LSB comment.
insserv: missing `Required-Start:' entry: please add even if empty.
pi@octopi:~ $

BillyBlaze, 2016-12-25:
oww and ofcourse remove ``$all`` 👍

Sodamaexchen, 2016-12-25:
ah,, wait... i removed the complete line... ups..

Sodamaexchen, 2016-12-25:
ok, now it got no error

BillyBlaze, 2016-12-25:
ok good :) now complete the last step in the wiki and see if it works 👍

Sodamaexchen, 2016-12-25:
i get the error:
pi@octopi:~ $ chromium-browser
/usr/bin/chromium-browser: line 138: lsb_release: command not found
/usr/bin/chromium-browser: line 139: lsb_release: command not found
/usr/lib/chromium-browser/chromium-browser: symbol lookup error: /usr/lib/chromium-browser/chromium-browser: undefined symbol: mmal_vc_init_fd
pi@octopi:~ $

Sodamaexchen, 2016-12-25:
so kernel upgrade ?

BillyBlaze, 2016-12-25:
Sadly, yes. This upgrade will be fixed in the next release of OctoPi.

Sodamaexchen, 2016-12-25:
ok, this will take a while....

Sodamaexchen, 2016-12-25:
YESSSS!! it works ! finally!

what can i do to thank you ? Case of Beer ?

BillyBlaze, 2016-12-25:
Awesome 👍 

I don't drink that often so, ehm, I guess just star this project 😄

closed 2016-12-25 by BillyBlaze

Sodamaexchen, 2016-12-25:
Done!