BillyBlaze/OctoPrint-TouchUI #80: Error "Service xserver has to be enabled to start service touchui"

iamaldovanucci, 2016-12-27:
Getting the error  Service ` Service xserver has to be enabled to start service touchui` when I try to run "sudo update-rc.d touchui defaults".

This is a fresh install of octopi on a pi 3. I can start the service manually by using "sudo service touchui start".

phredieFPV, 2016-12-27:
I also just ran into the same problem. Fresh install of raspbian jessie on an rpi 2.

BillyBlaze, 2016-12-27:
Thanks for reporting.

I have changed some code in the autostart repository.

Run:
```
cd ~/TouchUI-autostart/ && git pull
```

Afterwards redo step 8 and continue on, I'm pretty sure you also need to update your kernel to the latest version to get Chromium working, it seems like these issues are related to kernel changes.

iamaldovanucci, 2016-12-27:
I can confirm that the code you changed has corrected the issue. Thank you for your efforts, contributions to the community and lightning fast response!

phredieFPV, 2016-12-28:
Working perfectly now. Thanks Billy!

BillyBlaze, 2016-12-29:
Your welcome and again thanks for reporting!

I'll close this issue since it's fixed 👍

closed 2016-12-29 by BillyBlaze