BillyBlaze/OctoPrint-TouchUI #82: Webcam stopped showing up

mrp192, 2017-01-01:
So I accidentally uninstalled TouchUI and now I cant see my webcam. Any ideas on where to start looking to fix it?

mrp192, 2017-01-01:
Yes I have this is what it shows me. 
[image]

BillyBlaze, 2017-01-01:
Wow that is really weird. So, to get this right, you uninstalled TouchUI, installed it again, restarted everything, and then the controls appears within the webcam tab, without a webcam image?

mrp192, 2017-01-01:
I can be on whatever tab. Then if I click the webcam tab it just show the last tab I was on.

BillyBlaze, 2017-01-01:
Hmm, ok that indicates that something is going wrong with creating the webcam container.

Do you have other plugins installed beside the bundled ones?

mrp192, 2017-01-01:
yes

bizziebis, 2017-01-03:
I have this same issue, for me it never worked so I didn't pay much attention to it. But now I see this issue I suppose that it should actually work.

BillyBlaze, 2017-01-03:
Thanks for reporting, I really need some more information about installed plugins, os-, and browser versions.

bizziebis, 2017-01-03:
Just did so testing with disabling plugins. Turns out that in my case it's caused by [Slicer](https://github.com/kennethjiang/OctoPrint-Slicer) plugin. I use it always, that's why my webcam never worked with Touch UI,

BillyBlaze, 2017-01-03:
Ahh the Slicer plugin strikes again, its a great plugin, sadly it breaks on some certain browsers.

mrp192, 2017-01-03:
HAha so I just disabled slicer and that was it. I never use it. Pretty sure it had a update recently.

BillyBlaze, 2017-01-03:
ok since the Slicer plugin isn't 'officially' supported by TouchUI I will close this issue. I'll try to add support in the 1.0.0 version. (Also there is a new feature stop on the 0.x.x branch)

closed 2017-01-03 by BillyBlaze