BillyBlaze/OctoPrint-TouchUI #84: Stuck on Connecting to TouchUI Screen

lciscon, 2017-01-03:
I'm running Octoprint+TouchUI on a RPI3 7" Touchscreen in the kiosk mode.  I've been running it this way on several machines for over 6 months.  Suddenly I'm having an issue where it gets stuck on the "Connecting to TouchUI" screen.  I tried opening another window on that screen & I can manually connect to the sever on localhost.  So it should work.  I *think* this started when it auto-updated to the latest TouchUI 0.3.5 & Octoprint 1.3.0.  I'm not sure whether this is related to the other open bug about TouchUI not starting or not.  I tried those instructions to update the files & it didn't make a difference.  I'm going to try reverting to the previous version of TouchUI & Octoprint to see if that fixes it.

BillyBlaze, 2017-01-09:
Sorry I totally missed this ticket, first off thank you for reporting the problem!

Could you perhaps share your octoprint log? Your problem is pretty unique sofar, so no other solutions will help.

lciscon, 2017-01-11:
No worries.  I haven't had much time to diagnose it.  I temporarily at least avoided the issue by re-imaging my controller using a configuration that has an older version of TouchUI & octoprint and then manually updating octoprint to 1.3.0.  That worked.  I'll do some more testing and let you know.

lciscon, 2017-01-16:
Update on this:  I was incorrect when I said that I was able to work around it by manually updating to 1.3.0.  It was not actually updating.  I cannot figure out how to manually update octoprint without also updating TouchUI simultaneously.  When I update both of them the same thing happens every time.  And this has happened now on multiple machines.  Again I'm going from octoprint 1.2.15 to 1.3.0 and TouchUI 0.3.4 to 0.3.5.  I've attached my octoprint log.  Also FYI this behavior is only an issue on the local touch interface.  I can run TouchUI remotely just fine.

[octoprint.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/709112/octoprint.txt)

lciscon, 2017-01-16:
BTW the issue only appears once you've rebooted the machine.

BillyBlaze, 2017-01-16:
@foosel do you have any idea why the BlueprintPlugin is failing after updating OctoPrint to 1.3.0? It seems it simply cannot find my route at all. I cannot see anything weird in @lciscon log file. Can it be a faulty dependency?

The route it's ignoring:
https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/master/octoprint_touchui/api.py#L55

If there was something wrong with this, then I would've expected some more people reporting this issue 😄

BillyBlaze, 2017-01-16:
I think I found your issue, you're using an old version of the bootloader.

I think you beta tested the bootloader and we never updated it after the public release. (Where I changed the route `/plugin/touchui/echo` to `/plugin/touchui/ping`)

So to fix this follow these steps: https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Update-your-bootloader

lciscon, 2017-01-16:
Yes that fixed it.  But I had to do the full "install bootloader from legacy" instructions and delete the new TouchUI-autostart directory as well.  I've rebuilt it so many times that I must have had some legacy files intermixed.  Thanks for the help!!!

BillyBlaze, 2017-01-16:
Your welcome and wow at the pesky legacy code :) glad its working again!

closed 2017-01-16 by BillyBlaze