BillyBlaze/OctoPrint-TouchUI #85: [Request] Requesting removal of the "would you like to go fullscreen" popup

ntoff, 2017-01-07:
Requesting the "would you like to go fullscreen" popup be removed as per conversation on reddit.

Reason: Sometimes I use touchui's interface in a browser that doesn't have the ability to be clicked on (it just displays URL sources) and the popup is permanently there with no way to click "no" (or to click "yes" either) and since there's an option in the settings to go fullscreen, I feel the popup is no longer needed.

BillyBlaze, 2017-01-07:
Thanks for creating the ticket 👍 

As previously said, this legacy code can be removed due to "kiosk mode" on touchscreen devices. (Epiphany users have to find a workaround, e.g. install a script that execute F12, or install Chromium)

gretel, 2017-01-19:
@BillyBlaze that is a bit tricky - following http://raspberrypi.stackexchange.com/a/43336/8018 works fine for me.

BillyBlaze, 2017-02-05:
@gretel awesome, thanks for the info!

@ntoff I will remove this in the upcoming 0.3.6 release!

BillyBlaze, 2017-02-12:
This is fixed in the latest 0.3.6 release.

closed 2017-02-12 by BillyBlaze