BillyBlaze/OctoPrint-TouchUI #86: Unable to scroll on Adafruit 3.5" resisitive PiTFT

KTheMan, 2017-01-15:
My issue is identical to #62 where touch events are the only thing that are output to the interface, as opposed to scrolling movements. This leaves me unable to scroll through the interface. I have used the touch interface on the same exact hardware in the past, using Adafruit's instructions and custom kernel but haven't had any luck when using the 2016-03-18 OctoPi Jessie image and Touch UI autostart script.

I've already made sure that I can click, that it recognizes tough points and click-dragging, as well as starting over fresh twice. Not sure what wall I'm hitting. Any help is appreciated.

BillyBlaze, 2017-01-15:
Thanks for reporting, I got to find a more solid solution for this problem. Sadly the driver is sending touch events to Chromium but Chromium is telling us he is not using touch events. So we are listening to mouse events.

Anyhow; could you run `cat /proc/bus/input/devices` and post the output?

KTheMan, 2017-01-16:
What's funny is I did attempt to install the chromeTouch extension (supposedly it works well with kiosks for scrolling) through the browser that's brought up by TouchUI-autostart and couldn't install it with the chrome store saying that the browser wasn't a compatible version of chrome/chromium, but it did work when installing from a desktop environment.  FWIW, the plugin made no difference in either case. I don't know if this gives any insight into how it runs chromium differently from a different environment, but I thought it may give some information.

`I: Bus=0018 Vendor=0000 Product=0000 Version=0000
N: Name="stmpe-ts"
P: Phys=stmpe-ts/input0
S: Sysfs=/devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/stmpe-ts/input/input1
U: Uniq=
H: Handlers=mouse0 event1
B: PROP=0
B: EV=b
B: KEY=400 0 0 0 0 0 0 0 0 0 0
B: ABS=1000003
`
is the output returned.

BillyBlaze, 2017-01-16:
Thanks for the information.

Before I rattle on can we first make sure that you have indeed the same issue as #62.

1. Could you run:

    ```
    nano ~/TouchUI-autostart/chromium.xinit
    ```

2. Add ```--touch-events=enabled``` to the rest of Chromium parameters
3. Save (CTR-O, Enter, CTR-X)
3. Restart with `sudo service touchui restart`.
4. Check if scrolling is back

KTheMan, 2017-01-17:
That fixed it! Awesome! It immediately rendered a scroll bar, which hadn't
happened in any of my current or previous configurations.
I do have that same parameter elsewhere in chromium.xinit, under
TOUCHUI_ARGS as an 'if, then, else' statement. I ended up removing the
'then' conditional and changed the 'else' conditional to a 'then'
conditional and got the same result. For whatever reason touch events
weren't automatically detected under chrome even though it was recognized
as a TOUCHUI device

synman, 2017-02-03:
The problem is that there are different flavors of LCDs / touchscreens surfacing that the current chromium.xinit script can't detect.

For example I have a Sunfounder (https://www.sunfounder.com/rpi-accessory/monitor-screen-lcd/touch-screen/sf-3-5-lcdscr.html) and was having the same problem.

The right way to fix this is to see what is in your /proc/bus/input/devices and ensure your device's name is included in the pipe delimited string that gets grepped against.

For example I have a device name of:

Name="ADS7846 Touchscreen"

so my grep looks like:

TOUCHUI_DEVICE=$(egrep -o "ZH851|FT5406|ADS7846" /proc/bus/input/devices)

BillyBlaze, 2017-02-05:
@KTheMan I think I found a better way to solve this issue.

Could you perhaps run for me this command:
```
FRAMEBUFFER=/dev/fb1 & DISPLAY=:0.0 xinput --list --long | grep XITouchClass
```

And post the output here?

iammattcoleman, 2017-02-12:
@BillyBlaze I've got the Adafruit PiTFT Plus (https://www.adafruit.com/products/2441) that I'm using with a Raspberry Pi 2 Model B. I don't get any output from the command you asked @KTheMan to run.

Here's the full output of from xinput:
```
pi@octopi:~ $ DISPLAY=:0.0 xinput --list --long
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
	Reporting 4 classes:
		Class originated from: 6. Type: XIButtonClass
		Buttons supported: 10
		Button labels: "Button Unknown" "Button Unknown" "Button Unknown" "Button Wheel Up" "Button Wheel Down" "Button Horiz Wheel Left" "Button Horiz Wheel Right" None None None
		Button state:
		Class originated from: 6. Type: XIValuatorClass
		Detail for Valuator 0:
		  Label: Abs X
		  Range: 0.000000 - 4095.000000
		  Resolution: 0 units/m
		  Mode: absolute
		  Current value: 4095.000000
		Class originated from: 6. Type: XIValuatorClass
		Detail for Valuator 1:
		  Label: Abs Y
		  Range: 0.000000 - 4095.000000
		  Resolution: 0 units/m
		  Mode: absolute
		  Current value: 3805.000000
		Class originated from: 6. Type: XIValuatorClass
		Detail for Valuator 2:
		  Label: Abs Pressure
		  Range: 0.000000 - 255.000000
		  Resolution: 0 units/m
		  Mode: absolute
		  Current value: 0.000000

⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
	Reporting 3 classes:
		Class originated from: 4. Type: XIButtonClass
		Buttons supported: 10
		Button labels: "Button Left" "Button Middle" "Button Right" "Button Wheel Up" "Button Wheel Down" "Button Horiz Wheel Left" "Button Horiz Wheel Right" None None None
		Button state:
		Class originated from: 4. Type: XIValuatorClass
		Detail for Valuator 0:
		  Label: Rel X
		  Range: -1.000000 - -1.000000
		  Resolution: 0 units/m
		  Mode: relative
		Class originated from: 4. Type: XIValuatorClass
		Detail for Valuator 1:
		  Label: Rel Y
		  Range: -1.000000 - -1.000000
		  Resolution: 0 units/m
		  Mode: relative

⎜   ↳ stmpe-ts                                	id=6	[slave  pointer  (2)]
	Reporting 4 classes:
		Class originated from: 6. Type: XIButtonClass
		Buttons supported: 5
		Button labels: "Button Unknown" "Button Unknown" "Button Unknown" "Button Wheel Up" "Button Wheel Down"
		Button state:
		Class originated from: 6. Type: XIValuatorClass
		Detail for Valuator 0:
		  Label: Abs X
		  Range: 0.000000 - 4095.000000
		  Resolution: 0 units/m
		  Mode: absolute
		  Current value: 4095.000000
		Class originated from: 6. Type: XIValuatorClass
		Detail for Valuator 1:
		  Label: Abs Y
		  Range: 0.000000 - 4095.000000
		  Resolution: 0 units/m
		  Mode: absolute
		  Current value: 3805.000000
		Class originated from: 6. Type: XIValuatorClass
		Detail for Valuator 2:
		  Label: Abs Pressure
		  Range: 0.000000 - 255.000000
		  Resolution: 0 units/m
		  Mode: absolute
		  Current value: 0.000000

⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
	Reporting 1 classes:
		Class originated from: 3. Type: XIKeyClass
		Keycodes supported: 248

    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
	Reporting 1 classes:
		Class originated from: 5. Type: XIKeyClass
		Keycodes supported: 248

```

I manually installed the PiTFT with these instructions (https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/detailed-install) because the scripts included with OctoPi weren't working for me.

I am able to scroll in chromium when viewing TouchUI 0.3.5.

BillyBlaze, 2017-02-12:
I think I found a bug in 0.3.5

Could you guys with the ```stmpe-ts``` remove the ```--touch-events=enabled``` and upgrade to the latest TouchUI 0.3.6 release.

synman, 2017-02-12:
hmm.. I have the stmpe-ts now.  The Sunlander was junk.

You still want this procedure performed?  Seems like we just went through a whole exercise to get touch events working via chromium.xinit.

BillyBlaze, 2017-02-12:
Well I want it for you guys to work, without manually editing a file for it. Since X11 is reporting that XITouchClass is not used, then that means it might was another issue. Since I resolved another issue, I want to know for sure it wasn't caused by that one.

synman, 2017-02-12:
I have no ability to scroll with the following in chromium.xinit:

```
# Force Chrome to render with touch events
# Bugfix #46
TOUCHUI_DEVICE=$(xinput --list --long | grep XITouchClass)
if [ -z "$TOUCHUI_DEVICE" ]
then
        echo "TouchUI: touch events automatically detected"
else
        echo "TouchUI: touch events enabled"
#       TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
fi
```

synman, 2017-02-12:
and confirmed 0.3.6

[image]

BillyBlaze, 2017-02-12:
Ok, can you try this?:

1. ```sudo service touchui stop```
1. ```cd ~/TouchUI-autostart/```
1. ```git checkout master```
1. ```git pull```
1. ```sudo service touchui start```

synman, 2017-02-12:
testing now

synman, 2017-02-12:
works fine

iammattcoleman, 2017-02-13:
I upgraded to 0.3.6 and can still scroll. I am not using TouchUI-autostart; I manually enabled PiTFT support on OctoPi, installed xorg+nodm+matchbox-window-manager+chromium-browser, edited /etc/default/nodm to automatically log in as the pi user, then created /home/pi/.xsession:
```
#!/bin/bash
matchbox-window-manager -use_titlebar no -use_cursor no &

xset -dpms
xset s off

chromium-browser --kiosk "http://localhost"
```

BillyBlaze, 2017-02-22:
Okay, thank you all for your help! Glad it's fixed now.

closed 2017-02-22 by BillyBlaze

bolsoncerrado, 2017-04-05:
Howdy!

Just installed a PiTFT 3.5 and followed adafruit's tutorial to manually upgrade kernel and such....

I can see TouchUI fired on the miniTFT but I cant scroll.

I did the above:

pi@gerty:~ $ sudo service touchui stop

pi@gerty:~ $ 
pi@gerty:~ $ cd ~/TouchUI-autostart/
pi@gerty:~/TouchUI-autostart $ git checkout master
Already on 'master'
Your branch is up-to-date with 'origin/master'.
pi@gerty:~/TouchUI-autostart $ git pull
Already up-to-date.
pi@gerty:~/TouchUI-autostart $ sudo service touchui start

So Im assumiing im on the latest release already.

Anything else I should fix?

Thanks!

bolsoncerrado, 2017-04-05:
(not sure why some text appears striken above :??)

bolsoncerrado, 2017-04-05:
Hmmm the only differences I can see based on what I've read above is:

1)

Upon doing cat /proc/bus/input/devices

 I do get the same as KTheMan, EXCEPT he's getting:
S: Sysfs=/devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/stmpe-ts/input/input1

and I'm getting:
S: Sysfs=/devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/stmpe-ts/input/input0

2)

Contents of /boot/config.txt read:

dtoverlay=**pitft35-resistive**,rotate=270,speed=42000000,fps=20

however on adafruit's you can read:
dtoverlay=**pitft35r**,rotate=270,speed=42000000,fps=20

(I guess they just updated names but not screenshots on their tutorial?)

3) 

Contents of /etc/modules

stmpe_ts
stmpe-ts

Which is the correct one?!

BTW mine is a RPI3.