BillyBlaze/OctoPrint-TouchUI #87: current version not usable with epiphany-browser

gretel, 2017-01-19:
is epiphany supposed to be supported? the control and camera tabs are not usable. i'd send screenshots if helpful - using updated **octopi**. chromium works find - but slow.

BillyBlaze, 2017-01-21:
Epiphany is still supported, but slowly turning into deprecation if they do need keep up with the latest webkit engine.

Anyhow, I did check TouchUI in Epiphany and both the controls and the camera tab are working for me. Could you disable other plugins to make sure they're not the culprit. If they're not, I need some more information to reproduce your issue.

BillyBlaze, 2017-01-25:
Any information? If not, after 7 days this issue will be closed due to inactivity.

closed 2017-01-30 by BillyBlaze