BillyBlaze/OctoPrint-TouchUI #88: suggestion: show warning if "gcode visualizer" is enabled

gretel, 2017-01-19:
this really seems to drag down slow machines when enabled therefore i suggest showing a collapsable warning or similar.

BillyBlaze, 2017-01-21:
Thanks for the suggestion. We need to disable this plugin in touchscreen mode.

BillyBlaze, 2017-01-23:
So, Yeah, I was also here confused with the plugin "STL Viewer".

In touchscreen mode, the default gcode viewer will already be disabled.

BillyBlaze, 2017-01-23:
If it is not disabled in touchscreen mode then we might have found out your issue for #91 👍

gretel, 2017-01-23:
guess both ('gcode visualizer' and 'stl viewer') will impact performance and in relation to the size of the data. the latter one does not have a limiting option at the present time.

BillyBlaze, 2017-01-23:
Yeah, the latter is giving TouchUI a bit of an issue. I will definitely give a warning, or even better, disable STL viewer in touchscreen mode.

For now I have add'd this to checklist of #39

jstefanop, 2017-02-03:
seems like the gcode visualizer is still loading with current version of octoprint (1.3.1). What is the easiest way to completely disable the visualizer on the local device (i still want it to load on other devices that load ocroprint, the pi just cant handle loading it locally on chromium).

BillyBlaze, 2017-02-05:
Thanks, I can also confirm that the gcode visualizer currently loads in touchscreen mode.

I will fix this and make a new release.

BillyBlaze, 2017-02-12:
This is fixed in the latest 0.3.6 release.

closed 2017-02-12 by BillyBlaze