BillyBlaze/OctoPrint-TouchUI #91: [WEIRDO] Pi lagggggg....

bolsoncerrado, 2017-01-20:
Not sure who to blame here, but I'll try it here first since it mainly happens on TouchUI LOCAL pi+touch display....

Thing is, when prints finish overnight and printer stays idle for a long long time, raspi is lagggish as hell, and takes ages to respond to touch commands...

Happens on both pi2 and pi3. Restarting the service seems to solve the issue, so i wonder if there's some sort of memory hog increased by the idle status, like "terminal" log buildup on MEMORY instead than on "disk" cache, that makes Pi go slow as hell??

Obviously, Pi reacts fine when accessed remotelly via a computer in the LAN.

bolsoncerrado, 2017-01-20:
Perhaps related? https://github.com/foosel/OctoPrint/issues/1523

BillyBlaze, 2017-01-21:
Definitely not related.

Sounds like you have heavy memory or cpu usage.
Are you still using plugins like the Gcode visualizer?

bolsoncerrado, 2017-01-23:
Isnt gcode visualizaer coming by default with octoprint?

I dont particuarly use it although I dont discard having gone thru the tabs while printing perhaps?

BillyBlaze, 2017-01-23:
Ahh yeah, you're right. I was confused with the plugin "STL viewer".
The default GCode visualizer should be disabled in touchscreen mode.

I suspect there is a memory leak in javascript or that a plugin is just rendering too much data (e.g. visualizing)

Do you run any other plugins?

BillyBlaze, 2017-01-25:
Any information? If not, after 7 days this issue will be closed due to inactivity.

bolsoncerrado, 2017-01-25:
Well so far I posted a yt video on Octo github about a freezing I just suffered. as Gina puts it, Opi wasnt meant to run in local with local b rowser.... Perhaps Pi4 will help here ;)

closed 2017-01-25 by BillyBlaze