BillyBlaze/OctoPrint-TouchUI #92: Touch screen calibration help needed Everything else working great!!! Read closed issues.

jasonsaffle13, 2017-01-30:
I am sorry I looked through the closed section to try to find help and did see where there were numbered steps to try I followed them and I still am unable to get calibration to take. I am sorry for posting this here since the software does not seem to be the issue but I am unsure where to ask for help.

I added a raspberry pi to me Folger tech kossel printer running octiprint. I mounted a 3.5 touch screen display running touchui so I can control it with out logging into web page. The problem is my axis are all screwed up. I move left it goes up etc. I have tried the directions to calibrate that came with the touch screen: and tried changing inversion from 1 to 1.
[www.amazon.com]
[www.spotpear.com]

With no luck, I think that it is pulling the calibration from somewhere else but I have very little linux skills to follow what is going on. I plan on following this guide next:
[learn.adafruit.com]

Looking to see if anyone else has had this issue or if someone can point me to a forum that may be able to help more. Everything works over tablet, phone, web page, and even on the touch screen it's just not calibrated. The company has no clue since if I use there provide image or set it up from scratch with the provide drivers and calibration it works fine but since it is only in touchui they have no clue.

BillyBlaze, 2017-01-30:
Hello, I am afraid I won't be much of help to you with this one, your driver and hardware is totally different.

I would definitely try the adafruit tutorial.

synman, 2017-02-04:
@jasonsaffle13 you need to determine what your touchscreen's capabilities are.  The easiest way to do this is with xinput.

You need to list out all of your device's properties.  One of them will be for controlling x/y orientation for the touchscreen.

See here for some pointers:  https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=172025

For reference, I had to use the following (nothing that I tried via 99-calibration, Xsession, module load flags, etc worked:

`xinput --set-prop 'ADS7846 Touchscreen' 'Coordinate Transformation Matrix'  0 -1 1 1 0 0 0 0 1`

I found the best place to put this type of stuff is in TouchUI's chromium.init script if you are running it on octopi.  The window manager being used is very basic.  I'm sure you could install a more functional one, but for me at least, is NOT the direction I want to go.

Hope this helps

BillyBlaze, 2017-02-05:
I have updated the Wiki and created a script for launching the calibration tool.

Please read these instructions https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Troubleshooting#calibration

If anything is unclear, please let me know.

1sPiRe, 2017-02-06:
Hi
I'm a total noob with raspberry, but everything was going ok until I tried to calibrte it. I'm using a 7'' touch screen on a Pi3.
I followed your instructions, updated the bootloader....but when I reach the 4th step of the calibration, my screen turns off, and I've got no step to follow. It gives me some calibration values and then "If these values are estimated wrong, either supply it manually with the --precalib option, or run the 'get_precalib.sh' script to automatically get it ( through HAL)."
Is that the instructions to follow? and if yes how do I do that?
Sorry if it is a real newbie question....

synman, 2017-02-12:
Here's another good article on touch screen orientation and coordinate transformation matrix

https://wiki.ubuntu.com/X/InputCoordinateTransformation

I had to use this when I configured my Adafruit PiTFT Plus LCD/Touchscreen tonight with octopi.

The Adafruit touchscreen is 1000 times better than the eBay special I originally bought.

I had to insert the following into my chromium.xinit to get the touchscreen oriented properly:

`su $TOUCHUI_USER -c "xinput --set-prop stmpe-ts 'Coordinate Transformation Matrix'  0 1 0 -1 0 1 0 0 1"`

BillyBlaze, 2017-02-12:
Thanks for the link @synman .

After getting the latest updated, I can reproduce this problem. I have updated the bootloader where I include and run the file ```calibration.sh```. Here you can put your custom commands without losing your changes when upgrading the bootloader to the latest version.

I also changed wiki instructions @ https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Troubleshooting#calibration

1sPiRe, 2017-02-12:
Tks guys I'll try it asap, and give you feedbacks.

synman, 2017-02-12:
pulled maintenance down, created my calibration.sh shellscript and all is working fine.

closed 2017-02-25 by BillyBlaze