BillyBlaze/OctoPrint-TouchUI #95: Camera does not work -- local or network when in TouchUI mode

synman, 2017-02-04:
Following the instructions as provided on the wiki, including checking the URLs for m-jpeg streamer, I am not getting anything from my pi camera.

Works fine in full UI mode.  Any pointers regarding what I may have misconfigured?

synman, 2017-02-04:
[image]

And when I activate TouchUI using full Chrome on my MBP, I get this:

[image]

Chromium at least tries to do stuff but it appears on full chrome whatever js/css operation is being performed on #webcam is completely ignored.

If I have time I'll dig deeper into this myself today.  More than happy to help.

synman, 2017-02-04:
just went through the closed issues list... I see this may be related to the Slicer plug-in.... I have it installed and will disable it now to diagnose.

synman, 2017-02-04:
ok, so disabling slicer fixed the Camera view on full Chrome, i.e. my MBP, however it doesn't matter what I put in SnapStream's config (/webcam/?action=snapshot and /?action=snapshot) I cannot get a snapshot to render over chromium (LCD).

synman, 2017-02-04:
I guess you can close this... 

when using a fully qualified URL it works fine.  The only problem I see with this is that I'm currently relying on DHCP and on devices without bonjour (my android phone for example) I can't use the avahi hostname (octopi).

I plan on setting a reservation for the octopi at some point anyway so this is totally manageable for me.

BillyBlaze, 2017-02-05:
Hi, I'm glad you have figured this one out. I never completely understood OctoPi's security since I always built my own. Please do feel free to clarify any Wiki instructions.

closed 2017-02-05 by BillyBlaze