BillyBlaze/OctoPrint-TouchUI #96: xinput_calibration doesn't work after killing TouchUI service

SnowsongWolf, 2017-02-05:
I'm not very knowledgeable in things Linux but while following the calibration tutorial I found that I kept getting the "Cannot connect to x server" error message after killing the TouchUI service.  What confused me was it also killed the xinput --list command which was previously working fine.

What I determined was that xinput_calibration ran normally without killing TouchUI and generated the config values I needed.

BillyBlaze, 2017-02-05:
Thanks for reporting! Yes, you're spot on, xinput_calibration can no longer work if TouchUI is not running, these instructions became obsolete after Debian Jessie got released.

However I did now add a calibration start-up script, so X11 nows exactly to what instance it needs to attach itself. I have updated the Wiki, so this should be easier from now on.

closed 2017-02-05 by BillyBlaze

synman, 2017-02-05:
I would think this is because the window manager gets torn down when touchui shuts down... there's no xwindows session... no window manager, hence "cannot connect to x server".

fire up the matchbox window manager and try again.  I'm sure it'll work just fine.