BillyBlaze/OctoPrint-TouchUI #97: TouchUI and PSU Control do not play nicely together

synman, 2017-02-10:
I have also opened this as an issue in the PSU Control repo but there is currently a problem between TouchUI and PSU Control. When on a low density touchscreen (3.5 LCD attached to the Pi) with PSU Control and TouchUI active when you try to open the System/Settings/Logon spanner (see attached) the entire context menu is blanked out.

You can also simulate this fairly easily by just shrinking your browser's width (see attached).

[image]

synman, 2017-02-10:
Here's this issue for this on PSU Control:

https://github.com/kantlivelong/OctoPrint-PSUControl/issues/8

synman, 2017-02-10:
Here you can see what is going on....

[image]

BillyBlaze, 2017-02-11:
Thanks for reporting.

It's fixed in the maintenance branch and will be released a.s.a.p.

synman, 2017-02-11:
you rock!

closed 2017-02-12 by BillyBlaze

synman, 2017-02-12:
pulled maintenance down and the context menu is now displaying with proper dimensions.  Oddly, GCode Viewer has disappeared from my touchscreen display but is still visible on my phone.

Did it not make the cut?

BillyBlaze, 2017-02-12:
Yes, everything is like it should be. The GCode Viewer is enabled when you're no in Touchscreen mode. If you would like to enable Touchscreen mode for your mobile too, then enable it in the TouchUI Settings modal.

commit 3c0cdf71 (2017-02-11): :bug: Improve removal whitespace

Improve the removal of navbar items by looping through childNodes.

Fixes #97