BillyBlaze/OctoPrint-TouchUI #98: TouchUI won't boot on fresh OctoPi, RP3c and Official Foundation LCD

HCS-II, 2017-02-11:
Billy,

I have just made a fresh install of OctoPi 1.3, upgraded to 1.3.1, with TouchUI plugin installed on a RP3. I use an Official RP Foundation 7" touchscreen and followed to the letter (twice) the install instructions on your Wiki for the touchscreen. Both time, the Raspberry Pi still boots to the command line after a brief appearance of the black GUI screen (with mouse cursor in the middle).

Here is a copy of my Xorg.0.log file: [http://pastebin.com/nWsLTWtY](url)

Seems that Xserver closed because of unloadable "libinput" module.

Any idea what would cause the error? No steps I followed resulted in any error message.

Regards,

Mike

BillyBlaze, 2017-02-11:
Thanks for reporting. Okay something has been changed again. What if we install the required library that is missing with:

```
sudo apt-get install xserver-xorg-input-libinput
```

HCS-II, 2017-02-12:
Thanks for the reply!

System reports:
pi@OctoPrint:~ $ sudo apt-get install xserver-xorg-input-libinput
Reading package lists... Done
Building dependency tree       
Reading state information... Done
xserver-xorg-input-libinput is already the newest version.
xserver-xorg-input-libinput set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.

So it seems already installed. Reboot concludes the same way at CLI.

Mike

BillyBlaze, 2017-02-12:
Okay. After comparing your log with mine, I noticed that AIGLX was also throwing errors.

Let's try disabling that.

1. Run ```sudo nano /etc/X11/xorg.conf.d/99-calibration.conf```
1. At the end of the file paste the following:

```
Section "ServerFlags"
   Option "AIGLX" "off"
EndSection
```

1. Save (CTR-O, Enter, CTR-X)
1. Reboot

If not working afterwards, please re-share the Xorg.0.log

HCS-II, 2017-02-12:
We are getting somewhere...

There is no /etc/X11/xorg.conf.d directory and subsequently no 99-calibration.conf file.

The only place I could find an xorg.conf.d directory is under /usr/share/X11 but in it, boot sequence 99 is a file called 99-fbdev.conf (the one we created as per the Wiki).

Should I create the file or does that tell us there is something wrong in the Xserver install?

Mike

BillyBlaze, 2017-02-12:
Sorry, I meant add that section to the end of the file ```/usr/share/X11/xorg.conf.d/99-fbdev.conf``` (that we created in our wiki.)

HCS-II, 2017-02-12:
Paul, I added the commands and rebooted still ending at CLI.

Here is the new log file: [http://pastebin.com/fuNKbg82](url)

BillyBlaze, 2017-02-12:
Okay, sorry about this trail-and-error. I don't own the official Raspberry Pi Touchscreen so I cannot try this myself.

It is still loading GLX

Can you also add this section to the file?
```
Section "Module"
        Disable "dri"
        Disable "glx"
EndSection
```

and retry?

HCS-II, 2017-02-12:
Paul, still no dice.

Here is the new log file: [http://pastebin.com/ArSxAdET](url)

Mike

BillyBlaze, 2017-02-12:
Hmm no more errors...

Looks like something is failing outside xserver.

Can you try this and post the output of the second command (will be shown inside your commandline)

1. Run ```sudo service touchui stop```
1. Run ```sudo xinit ~/TouchUI-autostart/chromium.xinit```
1. Press CTR-C to exit afterwards

HCS-II, 2017-02-12:
Here you go: [http://pastebin.com/uLqLPF6D](url)

The debut closed by itself.

Mike

BillyBlaze, 2017-02-12:
Thanks!

Looks like Chromium crashed after not finding lsb-release, what if we install that one with:
```sudo apt-get install lsb-release```

And you retry?

HCS-II, 2017-02-12:
Here the result: [http://pastebin.com/pbnrtE8Y](http://pastebin.com/pbnrtE8Y)

After reboot, we're still at CLI.

BillyBlaze, 2017-02-12:
No cigar 😞 thanks... Let's try something else. Maybe we need to remove ```--no-first-run``` from the chromium parameters.

1. Run ```nano ~/TouchUI-autostart/chromium.xinit```
1. At line 36 remove ```--no-first-run ```
1. Save & Exit (CTR-O, Enter, CTR-X)
1. Rerun ```sudo xinit ~/TouchUI-autostart/chromium.xinit```

HCS-II, 2017-02-12:
Still at CLI after reboot.

Here is the last xinit debug: [http://pastebin.com/U2kX9XhL](http://pastebin.com/U2kX9XhL)

BillyBlaze, 2017-02-12:
Bizar... Lets dive a bit deeper in the Chromium log.
Can you enable that?

1. Run ```nano ~/TouchUI-autostart/chromium.xinit```
1. At line 9 remove the hashtag ```#TOUCHUI_DEBUG="--remote-debugging-port=9222 --enable-logging --v=100"```
1. Save & Exit (CTR-O, Enter, CTR-X)
1. Rerun ```sudo xinit ~/TouchUI-autostart/chromium.xinit```
1. Share the new log found at ```/home/pi/.config/chromium/chrome_debug.log```

HCS-II, 2017-02-12:
Here we are:

pi@OctoPrint:~` $ cat /home/pi/.config/chromium/chrome_debug.log
[979:979:0212/115158:VERBOSE1:breakpad_linux.cc(1838)] Breakpad disabled
[1:1:0212/115158:VERBOSE1:sandbox_linux.cc(152)] Lacking support for seccomp-bpf sandbox.
[1:1:0212/115158:VERBOSE1:zygote_main_linux.cc(598)] ZygoteMain: initializing 0 fork delegates
[1:1:0212/115158:VERBOSE1:zygote_main_linux.cc(314)] Unable to load plugin /usr/lib/chromium-browser/libpepflashplayer.so /usr/lib/chromium-browser/libpepflashplayer.so: cannot open shared object file: No such file or directory
[1:1:0212/115158:INFO:systeminfo.cc(82)] Available number of cores: 4
pi@OctoPrint:~ $ 

Mike

BillyBlaze, 2017-02-12:
Thanks!!! Sigh, another dead-end.. These startup errors are the same as mine, but it crashes silently..

Can you try this?

1. Run ```nano ~/TouchUI-autostart/chromium.xinit```
1. At the end of the file change ```su $TOUCHUI_USER -c "chromium-browser $(echo $CHROME_ARGS|tr -d '\n')"``` into ```su $TOUCHUI_USER -c "chromium-browser"``` 
1. Save & Exit (CTR-O, Enter, CTR-X)
1. Rerun ```sudo xinit ~/TouchUI-autostart/chromium.xinit```
1. Share the new log found at ```/home/pi/.config/chromium/chrome_debug.log```

HCS-II, 2017-02-12:
Seems like it's exactly the same:

pi@OctoPrint:~ $ cat /home/pi/.config/chromium/chrome_debug.log
[979:979:0212/115158:VERBOSE1:breakpad_linux.cc(1838)] Breakpad disabled
[1:1:0212/115158:VERBOSE1:sandbox_linux.cc(152)] Lacking support for seccomp-bpf sandbox.
[1:1:0212/115158:VERBOSE1:zygote_main_linux.cc(598)] ZygoteMain: initializing 0 fork delegates
[1:1:0212/115158:VERBOSE1:zygote_main_linux.cc(314)] Unable to load plugin /usr/lib/chromium-browser/libpepflashplayer.so /usr/lib/chromium-browser/libpepflashplayer.so: cannot open shared object file: No such file or directory
[1:1:0212/115158:INFO:systeminfo.cc(82)] Available number of cores: 4

BillyBlaze, 2017-02-12:
Argh... Anything useful in any of the generic logs?

```cat /var/log/syslog | tail -30```
```cat /var/log/messages | tail -30```
```cat /var/log/kern.log | tail -30```

HCS-II, 2017-02-12:
Here are the results: [http://pastebin.com/X8kiQn1j](http://pastebin.com/X8kiQn1j)

BillyBlaze, 2017-02-13:
What if you run ```chromium-browser``` what is the output you get? Maybe we can get the error from there.

HCS-II, 2017-02-13:
As expected I get an error and here is the detail: [1725:1725:0213/173345:ERROR:browser_main_loop.cc(271)] Gtk: cannot open display:

Would it be better if I started with a full Rapsbian Pixel install?

Mike

mevdschee, 2017-02-13:
Paul asked me to help. I'll try.

>Gtk: cannot open display:

Are you logged in via SSH? If so, first run:

```
export DISPLAY=':0.0'
```

On my machine I can debug Chromium by running:

```
chromium-browser --debug
```

Type 'r' at the gdb prompt to run and 'q' to quit.

HCS-II, 2017-02-14:
Thanks for helping out! 

Yes, until I get TouUI GUI working, I access via SSH.

After having executed the display command, here is the result of running Chromium:
[1790:1790:0213/194845:ERROR:browser_main_loop.cc(271)] Gtk: cannot open display: :0.0

With startx, run from the RPI directly, I get only a black screen.

Here is the result of the Chromium debug:
pi@OctoPrint:~ $ chromium-browser --debug
# Env:
#     LD_LIBRARY_PATH=/usr/lib/chromium-browser:/usr/lib/chromium-browser/libs
#                PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
#            GTK_PATH=
# CHROMIUM_USER_FLAGS=
#      CHROMIUM_FLAGS=  --enable-pinch --ppapi-flash-path=/usr/lib/chromium-browser/libpepflashplayer.so --ppapi-flash-args=enable_stagevideo_auto=0
/usr/bin/gdb /usr/lib/chromium-browser/chromium-browser -x /tmp/chromiumargs.gdhul4
GNU gdb (Raspbian 7.7.1+dfsg-5+rpi1) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/lib/chromium-browser/chromium-browser...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/lib/chromium-browser/chromium-browser --enable-pinch --ppapi-flash-path=/usr/lib/chromium-browser/libpepflashplayer.so --ppapi-flash-args=enable_stagevideo_auto=0 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x74888290 (LWP 1858)]
[New Thread 0x74088290 (LWP 1863)]
[1855:1855:0213/195422:ERROR:browser_main_loop.cc(271)] Gtk: cannot open display: :0.0
[Thread 0x74888290 (LWP 1858) exited]
[Thread 0x754ad000 (LWP 1855) exited]
[Inferior 1 (process 1855) exited with code 01]
(gdb) 

This debug is essentially the same via SSH or directly on the RPI.

Mike

mevdschee, 2017-02-14:
@Mike, It is clear that you have no active X session and without X running you cannot run Chromium. Have you ever got a graphical interface running? Or was it never working?

HCS-II, 2017-02-14:
Maurits,

No, I never got a GUI operating. When the RPI boots, I briefly get a black GUI screen with a mouse cursor in the middle and then it returns to CLI.

I started with a fresh OctoPi install (CLI, without GUI) and followed the Wiki's [instructions](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)) and in Step 4, I installed Xserver with the following command:
sudo apt-get install --no-install-recommends xinit xserver-xorg xserver-xorg-video-fbdev x11-xserver-utils matchbox unclutter chromium-browser

Would I maybe missing a package? 

Mike

mevdschee, 2017-02-14:
Not sure this is related:

https://forums.kali.org/showthread.php?32745-Rpi3-Black-screen-and-cursor

But it seems.

HCS-II, 2017-02-14:
I don't seem to have lightdm installed so this may be the cause. I installed it and still no dice. Then I tried slim without success

Would the command with --no-intall-recommends have kept xserver install from fetching lightdm? I'll try the install without this exclusion command.

Mike

mevdschee, 2017-02-14:
Is the disk space sufficient? I guess it is..  as you can install packages. The "--no-intall-recommends" is put there intentionally. Nevertheless you could try. I am also clueless at the moment.

HCS-II, 2017-02-14:
Just tried re-running the apt-get install without the argument and no further install took place.

As for memory, I have 12GB left so no problems on that side.

What I'll do next is a clean full Jessie install, add OctoPrint, activate TouchUI plugin and then run the Wiki's intructions for a full Jessie install.

Regards,

Mike

BillyBlaze, 2017-02-14:
Thanks for all the feedback and your patience @HCS-II 
And Thanks @mevdschee  for all the help!!

I have all the confidence that installing Jessie Full will work, since xserver is already pre-configured. Sadly I do want to know what is the cause for this. Since we have alot of Offical Raspberry Pi Touchscreen users.

HCS-II, 2017-02-14:
O.K., I've just done a clean full Jessie install, loaded OctoPrint from GitHub (lots of work and still need to setup webcam and streaming service), enabled TouchUI plugin and followed the Wiki instructions. I often checked that startx enabled the GUI. Result: success!

Now, I would also like to know what doesn't work for the install on OctoPi as it's a lot less work. 

If you know of others with a Official raspberry lcd, maybee you could ask one of them to try a fresh OctoPi install?

On my side, I'll try again the OctoPi install from scratch. One I have installed xserv packages, should I be able to boot into the GUI using the startx command? Because I think that's the fundamental issue.
 
Regards,

Mike

BillyBlaze, 2017-02-14:
Ok 👍 Glad that it is working!

I am pretty sure there is something wrong with Chromium and the Official touchscreen. The Official Touchscreen always get the newests features, so it might auto-enable something within Chromium that makes it crash. e.g. hardware acceleration or something similair.

For example, if xserver closes chromium-browser then it would still throw an error like this:
```
[ERROR:chrome_browser_main_extra_parts_x11.cc] X IO error received (X server probably went away)
```

However, after looking into the parameters we could enable Chrome to trace everything to a file and I can then view the stacktrace of Chrome itself.

PhilMaddox, 2017-02-22:
Hi guys . Still looking for a tester ???  I have a Pi3 ( and pi2) and the official 7 inch LCD . I would really like to get this working for my own project. Not really at home with Linux but understand enough to help i think or at the very least follow instructions 
From what i've read above i should do a clean full Jessie install... then install Octopi ...then touchui addon and then ????

HCS-II, 2017-02-22:
Well we know the "long" route works (full Jessie, manual install of OctoPrint, etc...).

What needs to be confirmed is the "short" route via the OctoPi image. If you want to help, just do a clean OctoPi install then follow the instructions in the Wiki for OctoPi. 

In the next week, I plan on re-doing a clean install with OctoPi and try to see if I can debug it. What'll do is plug the RPi directly to the Ethernet as I noticed that some `apt-get` via my Wi-Fi connection would sometimes fail.

Regards,

Mike

BillyBlaze, 2017-02-22:
Hi @PhilMaddox,

Thanks for providing your help with this issue. I just got word that I will receive a RPI Touchscreen device one of these days. So I think it would be wise if I first try to reproduce this problem on my own. This will save us going back and forward through diving into log files.

jbdiablo, 2017-02-23:
This issue may be fixed.  I just bought the 7" touchscreen and did the steps in the guide over a fresh install of octopi and everything works.  TouchUI comes up on the screen.  Only issue I have left is to figure out an on screen keyboard so I can login to octopi and be able to input temps and gcode in the terminal.

jbdiablo, 2017-02-23:
I did get the error after launching chromium browser but followed the guide to update the kernel and everything worked fine after that.

Invarti, 2017-02-25:
Just echoing the feedback above by @jbdiablo the wiki instructions are spot on and working with the official 7" Pi touchscreen. The lack of an on-screen keyboard is an issue. Any recommendations?

jbdiablo, 2017-02-25:
I noticed the option for the on-screen keyboard in the TouchUI settings about 10 minutes after I posted here.  

Only issue I see now which I have not seen mentioned anywhere is the camera does not show up in controls in the standard UI or TouchUI on the official touchscreen.  Not really an issue or on purpose since the touchscreen is right next to my printer.

Invarti, 2017-02-25:
A few days ago I came across a forum post mentioning that very issue. I ignored it as my Pi will be positioned next to the printer, so I can view the print directly with my Mk1 eyes.

BillyBlaze, 2017-02-25:
Thank you everyone for the information. I just received my own RPi Touchscreen and I can also confirm that the issue has been resolved in Chromium 👍 

I do need to fix that the virtual keyboard is no longer automatically activated on a touchscreen device. I will address this in the upcoming release.

closed 2017-02-25 by BillyBlaze

PhilMaddox, 2017-02-25:
Yup got mine installed to . No issues and nice and fast . I dont have it auto booting to the touchUI yet thou. I noticed in the tutorial that the initial part installs Chromium yet in the clean full version of Jessie with Pixel CHromium is installed by default.  Bit of help in how to boot straight to the TouchUI interface would be appreciatted or do i simply follow the  https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(Jessie-Full) 
Thanks