BillyBlaze/OctoPrint-TouchUI #99: ToucUi not scrolling

sajovicd, 2017-02-12:
Hello,
I have strange problem. 
When I installed TouchUI by your great tutorial everything was working great, but now after aproximately month of use touchui, I cant scroll through the page, so I am stuck with inital screen because cant scroll to top of page to select some other screen. I use original PI 7" screen from adafruit (it realy cant comper over ebay cheap alternatives).
If I press stop button for printing, I got popup window asking to confirm then I can slide but just until I select another screen.
Also TouchUi somethimes crashes, I have this strange behaviour for a week now. 
I was searching for solution but no luck.
What can I do to get back slscrooling?

BillyBlaze, 2017-02-12:
Hello,

Have you upgraded TouchUI to 0.3.6?

sajovicd, 2017-02-13:
Hello,

Sorry for late replay.
After upgrade it is working again.

Thank you very much, and thanks for the great work.

BillyBlaze, 2017-02-13:
No problem 👍 Awesome to hear it's working again.
Please post any issue if you experience an issue.

closed 2017-02-13 by BillyBlaze