BillyBlaze/OctoPrint-TouchUI #101: No G-Code Viewer only on Pi Screen

cosmith71, 2017-02-13:
First of all, thanks for the work.  It is greatly appreciated.

I'm having a strange issue with TouchUI.  I'm running it on a RPi 3 with the official 7" touchscreen.  On this touchscreen, the G-Code Viewer does not appear in the menu.  Everything else seems to be working OK.  If I log on with any other device, it's there.  Any suggestions?

Thanks,

--Colin

BillyBlaze, 2017-02-13:
Your welcome 👍 The G-code Viewer is disabled on Touchscreen devices for performance. You could go TouchUI Settings and disable Touchscreen mode, however it is highly recommended not do this because the Gcode-viewer might/will crash your Chromium and interfere with print job. Depending on the size of the Gcode.

cosmith71, 2017-02-13:
Ah.  I'm happy to report it is working perfectly then!  No big deal, I just noticed it was missing and had been there before, so I was wondering what I broke.  ;)

BillyBlaze, 2017-02-13:
Awesome :) Also thanks for reporting, please feel free to add any issue if needed.

closed 2017-02-13 by BillyBlaze