BillyBlaze/OctoPrint-TouchUI #102: No TouchUI on epiphany-browser on localhost?

Eddiiie, 2017-02-20:
Hi,
I swear it was there just a few hours ago and no matter what I cannot get the UI back or the icon that appears next to the announcements icon at the top.

If I change to the chrome browser, TouchUI opens right away.   

??

BillyBlaze, 2017-02-20:
Have you might installed a new plugin the last couple of days? It sounds like TouchUI is stumbling upon a javascript error.

You haven't upgraded TouchUI or Epiphany lately?

Eddiiie, 2017-02-21:
This is a build from scratch.  Obviously something changed, I know it was working; I see how-to's using the epiphany browser.  I think it is related to screen resolution, although it is a 3.5 inch screen, the scroll bars that appear on the epiphany browser window are scroll forever.    If only chrome didnt use so much CPU..

Any clues I can look for?  You mention javascript, perhaps I disabled javascript ?   Would that stop the icon from appearing at the top, too??

BillyBlaze, 2017-02-21:
Well unless you run the old models, you do have a multiple core processor.

From my experience Epiphany-browser is a bit 'janky' while Chromium is way smoother. Also Chromium uses touch events these days while Epiphany is still using mouse events (which TouchUI emulates into touch events)

Anyhow back to the subject; could you perhaps share you `~/.octoprint/logs/octoprint.log`?

Eddiiie, 2017-02-21:
Hi thanks :)

The localhost's IP is 192.168.254.73 fyi...

```
pi@toyrep:~/.octoprint/logs $ cat octoprint.log
2017-02-21 11:23:32,896 - octoprint.plugins.announcements - INFO - Loaded channel _releases from http://octoprint.org/feeds/releases.xml in 0.21s
2017-02-21 11:23:34,327 - octoprint.plugin - ERROR - Error while calling plugin stats
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugin/__init__.py", line 229, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 515, in on_event
    self.statDB.execute(sql)
AttributeError: 'StatsPlugin' object has no attribute 'statDB'
2017-02-21 11:23:34,676 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.0-RCBugFix (Github)"
2017-02-21 11:23:34,998 - octoprint.plugins.announcements - INFO - Loaded channel _news from http://octoprint.org/feeds/news.xml in 0.2s
2017-02-21 11:23:35,098 - octoprint.plugins.stats - INFO - Printer Stats
2017-02-21 11:23:35,172 - octoprint.plugins.stlviewer - INFO - STL Viewer loaded!
2017-02-21 11:23:35,255 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.254.73/'}
2017-02-21 11:23:58,241 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.254.73:5000/'}
2017-02-21 11:24:05,188 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://localhost/'}
2017-02-21 11:24:05,436 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://localhost:5000/'}
2017-02-21 11:24:12,012 - octoprint.server.util.sockjs - INFO - New connection from client: ::1
2017-02-21 11:24:28,463 - octoprint.util.pip - INFO - Using "/home/pi/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2017-02-21 11:24:28,465 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2017-02-21 11:24:39,058 - octoprint.util.pip - INFO - pip installs to /home/pi/OctoPrint/venv/lib/python2.7/site-packages/, --user flag needed => no, virtual env => yes
2017-02-21 11:24:53,496 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-02-21 11:26:25,512 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.254.65
2017-02-21 11:26:27,719 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from http://octoprint.org/feeds/octopi.xml in 0.19s
2017-02-21 11:26:27,850 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from http://plugins.octoprint.org/feed.xml in 0.1s

```

```
pi@toyrep:~/.octoprint/logs $ ls -l
total 528
-rw-r--r-- 1 pi pi   2743 Feb 21 11:26 octoprint.log
-rw-r--r-- 1 pi pi 258915 Feb 20 01:43 octoprint.log.2017-02-20
-rw-r--r-- 1 pi pi  34816 Feb 20 01:32 octoprint_stats.db
-rw-r--r-- 1 pi pi      0 Feb 18 15:57 plugin_cura_engine.log
-rw-r--r-- 1 pi pi 223810 Feb 19 11:20 plugin_pluginmanager_console.log
-rw-r--r-- 1 pi pi      0 Feb 18 15:57 plugin_softwareupdate_console.log
-rw-r--r-- 1 pi pi      0 Feb 18 15:57 serial.log
pi@toyrep:~/.octoprint/logs $

```

BillyBlaze, 2017-02-22:
Hmm, interesting. thanks for the log. Seems like TouchUI is not starting up at all. It looks like the plugin 'Printer Statistics' is throwing an error and maybe preventing TouchUI from starting.

Could you disable that plugin, restart octoprint and see if TouchUI is still not starting up?

BillyBlaze, 2017-02-25:
Any information?

Eddiiie, 2017-02-25:
Hi,
I disabled  printer stats and it had no impact, so I thought I'd try disabling everything except the default and touchui  and now it works again on the epiphany browser.   Cool.   Now to re-enable one by one.

Eddiiie, 2017-02-25:
Plugin - OctoPrint-Slicer-0.9.6    from the plugin manager it is "Full-featured Slicer"
Enabling this plugin stops touchui from working (?) on the epiphany browser for some reason.
I uninstalled it, will test some more later.

BillyBlaze, 2017-03-15:
Thanks for details 👍 

The "Full-featured Slicer" plugin strikes again. I should do something with that.

closed 2017-03-27 by BillyBlaze