BillyBlaze/OctoPrint-TouchUI #103: Misplaced texts on Raspbberry Pi HDMI screen

trgtylcnky, 2017-02-20:
Hi,
I am using touch screen on Raspberry Pi. While using TouchUI, some texts are placed in ridiculous positions.
For example, when I am in camera tab:
[image]

Or I am in  main menu, this is not permanent, disappears when I use slider.
[image]

Do you suggest anything for me to investigate this issue?

BillyBlaze, 2017-02-20:
Thanks for reporting. 

It seems like TouchUI is having a javascript error and is therefor failing to start-up.
Have you installed additional plugins?

trgtylcnky, 2017-02-21:
Thanks for replying.
No additional plugin is installed. I realized that in one of my previous installation it was working, I was updated the whole raspbian. Are there any specific package I should upgrade do you think? I thought about epiphany-browser but it is already on the newest version. 
I prefer not to update whole system this time because it takes so long and I am really sorry for saying this.

BillyBlaze, 2017-02-21:
Could you share your `~/.octoprint/logs/octoprint.log`?

If there is no information then I need to know how you're starting epiphany so you can get the javascript console messages 👍

trgtylcnky, 2017-02-24:
Sorry for late reply, but I couldn't managed to observe the same bugs. The reason is that other problems are occurred on my system and I had to reinstall everything. I am now running OctoPi image with latest versions and problem seems solved. 
Though I am still observing random problems such as kernel panic or being unable to slide TouchUI panel (random times, after one print is done etc. ), I think this issue can be closed.

BillyBlaze, 2017-02-25:
Alright, glad you where able to fix this issue!

closed 2017-02-25 by BillyBlaze