BillyBlaze/OctoPrint-TouchUI #104: Error running chromium-browser

arb1701, 2017-02-21:
I have followed the install instructions for the OctoPrint-TouchUI-autostart and I am getting the following error at step 10:

pi@PI3-IIIP:~ $ chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0220/232837.425124:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[1882:1882:0220/232837.431265:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[1882:1882:0220/232837.439719:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display:

Any help you give would be greatly appreciated.

--Allan

BillyBlaze, 2017-02-21:
Thanks for reporting.

The error eventually is `Gtk: cannot open display:` and is good. What happens when you reboot? Does TouchUI appear?

If not could you please share some information about which touchscreen you're using and if you're running OctoPi, Jessie Light or Jessie Full?

arb1701, 2017-02-21:
The touchscreen is the Eleduino 7" IPS (1024-600) connected via HDMI to my PI3 and it is working properly as I see all of the messages on the screen as the Pi boots. I am running OctpPi Jessie Light. When the Pi boots, it goes straight to the console log in. I never see any attempts to open TouchUI; i.e. black screen, etc.

When I originally did the Octoprint-TouchUI-autostart, I did get the error messages which indicated that my kernel needed to be upgraded, and I followed the steps to do that. I got the errors I posted after the kernel upgrade. I have tried forcing a reinstall of xserver and chromium by adding the -f flag to the install command. I have also tried removing and reinstalling, but I still get the same error messages.

I also did apt-get update and apt-get upgrade...still no joy.

I looked in /usr/lib/chromium-browser and I see nacl_helper_bootstrap and nacl_helper_nonsfi, but no nacl_helper.

arb1701, 2017-02-21:
I found the problem. I ran startx and it failed. Checking the log, it wasn't detecting the display at fb1. I changed to fb0 and it's now working. The only issue I now see is that the webcam isn't showing in the camera window on the touchscreen. I connected to OctoPrint on my laptop and I can see the webcam so I know it's working. It;s not a huge deal since I'm right at the printer while running on the touchscreen.

I should have tried startx before posting this issue, but the errors before the Gtk: cannot open display had me thinking there was a serious problem.

Thank you for your time. I really like OctoPrint/OctoPi.

--Allan

arb1701, 2017-02-21:
Got the webcam working...I found in closed issues where the full url needs to be entered in the webcam settings. Working like a champ now.

--Allan

BillyBlaze, 2017-02-22:
Hi Allan,

Awesome! I'm glad you where able to figure out this problem 👍 I should really clarify on using the entire url in the troubleshooting section for the webcam.

I am wondering if my wiki is unclear about using fb0 with a HDMI device, is there room for improvement?

Thanks,
~Paul

closed 2017-02-25 by BillyBlaze