BillyBlaze/OctoPrint-TouchUI #105: "Touchscreen mode" does not appear on touch screen

trgtylcnky, 2017-02-24:
Hi,
I followed these instructions: https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

I am using OctoPi 0.13 and everything is up-to-date.
TouchUI boots without any problem but in "TouchUI Settings" menu, there is no option of "Toggle touchscreen mode" so I cannot slide the page.
When I am remotely connected from PC, this option exists.
Thanks,

BillyBlaze, 2017-02-25:
Hi,

Thank you for reporting. I can confirm that this option only appear when the device is in muis-mode. And the latest Chromium is using touch-mode.

I will fix this in the upcoming release.

trgtylcnky, 2017-02-25:
Thank you very much. Can I ask when will the patch  be ready? Or can you suggest another method (such as using another browser) until the patch ready.

BillyBlaze, 2017-04-16:
This has been fixed in the latest 0.3.7 release.

closed 2017-04-16 by BillyBlaze