BillyBlaze/OctoPrint-TouchUI #106: Login not remembering

SYC-O, 2017-02-27:
I have found that no matter what I do every time I login to octoprint on my phone or tablet via the touch ui I have to enter my login details. I've checked the remember details checkbox and yet it still forgets it. Also tried switching to normal view and logging in and then back to the touchui but still forgets my details.

bzeeman, 2017-03-06:
Same problem here.

BillyBlaze, 2017-03-07:
Hello,

Thanks for reporting. If you do not provide any information about the devices, browser or any other information, then I can not look into this bug.

I am going to assume you both are on an iOs devices and this bug is already reported in the main repository foosel/OctoPrint/issues/1792

BillyBlaze, 2017-03-15:
Closed due to inactivity. Please feel free to open a ticket if you can provide me some more details.

closed 2017-03-15 by BillyBlaze