BillyBlaze/OctoPrint-TouchUI #107: Webcam feed resize to fit screen

aphrroman, 2017-03-09:
I was wondering if there is a way to resize the webcam feed to fit the screen. Currently it is too big and scrolling is necessary. Also I have no option to select the color of the interface through TouchUI settings. Not a big deal but I'm a bigger fan of purple than blue. Lol
Current setup
OctoPrint 1.3.1
TouchUI 0.3.6
Rpi webcam rev 1.3
Official Rpi 7" Touchscreen

BillyBlaze, 2017-03-15:
Hello,

Thanks for suggesting. Sadly it is not possible to make the webcam fit the screen in both width and height. The image will loose it's aspect ratio if we fiddle with height and width. Also in addition if we would resize the image in height on a 3.5"  touchscreen device then the image would get to small.

I suggest setting up your webcam in 19:6 aspect ratio (see [RPi docs](https://www.raspberrypi.org/documentation/usage/webcams/) for examples) since that would match the aspect ratio of your touschreen device.

Also you cannot change the colors in the `TouchUI settings`, but these settings can be found in `OctoPrint Settings > TouchUI`

BillyBlaze, 2017-04-16:
After fixing this issue in my other plugin OctoPrint-Fullscreen. I could also use the CSS property `object-fit: contain;` in TouchUI.

So I am re-opening this feature request and implement it in the next version.

BillyBlaze, 2017-04-16:
This has been fixed in the latest 0.3.7 release.

closed 2017-04-16 by BillyBlaze