BillyBlaze/OctoPrint-TouchUI #109: Control tab blank on Pi2 with Adafruit TFT3.5

BRLewitt, 2017-03-11:
Running the latest OctoPi image on a Pi2 with the Adafruit TFT setup per the boot to browser instructions in the Wiki.  I'm using the 3.5" screen and got everything working (even the webcam) but the body of the Control tab is blank... no arrows, home buttons, or fan controls.

mschraeder, 2017-03-14:
Sorry for the obvious question, but are you sure you're logged in?

BRLewitt, 2017-03-14:
Yup.  I had installed it from a zip file and I confirmed I was logged in.  I uninstalled TouchUI and reinstalled from the OctoPi plugin manager with no other plugins installed and now it's working as expected.  I have to determine what plugins break the webcam, but that's another issue and was the reason why I'd installed from a ZIP to begin with as a code tweak was suggested in another thread to address the webcam not showing images.  So this issue can be closed.  Thanks.

BillyBlaze, 2017-03-15:
Hello,

Thanks for reporting and luckily this was not a bug 👍 That would have been a major one!
Please feel free to post any issues if you encounter them.

closed 2017-03-15 by BillyBlaze