BillyBlaze/OctoPrint-TouchUI #112: Control interface formatting misaligned

leftyfb, 2017-03-15:
After following the instructions to fix issue #81 , the interface shows on boot but the "control" interface is all
misaligned and not really usable without inadvertently clicking on the wrong control.

[image]

BillyBlaze, 2017-03-15:
Hello,

Thanks for reporting.

I think I see you're using - but correct me if I am wrong - the plugin "M33 Fio"?
Currently both plugins do not support each other and I am afraid that this is caused by this plugin.

leftyfb, 2017-03-15:
I've been using both plugins together for months now. It was just after these updates that this broke. Do you mean to tell me that because of my 3D printer, I cannot use TouchUI like I've been doing since last summer? I think I notice that most of the issue might be because of the overly large footer at the bottom.

BillyBlaze, 2017-03-15:
I understand your frustration, but you probably updated Chromium and that is now rendering this new issue. You could downgrade Chromium to the version where it was work-able for you.

The main issue is that support for this plugin requires a lot of work and time, and the latter is something I don't have laying around.

I will take a look at fixing some of the major parts of the Controls, but down-grading Chromium is a more quicker fix atm.

leftyfb, 2017-03-15:
I just downgraded to chromium-browser_51.0.2704.91-0ubuntu0.14.04.1.7015 from chromium-browser_56.0.2924.84-0ubuntu0.14.04.1.1000. No change. This was working in November when the older version was installed.

leftyfb, 2017-03-15:
I have now tried all 4 versions of chromium since it's release on the Pi in September. All are experiencing the same issue. I do not think this is related to the version of chromium. Any other ideas I can try? I'm going to be displaying this as a demo during a "Raspberry Jam" event on Saturday. Not having the pi controlling my printer would be pretty pointless.

BillyBlaze, 2017-03-15:
Hmm, that is pretty bizar. I have absolutely no idea what else have changed with apt-get upgrade since your last working state. I think there is an upgrade log somewhere in `/var/log` but I am afraid this is going to be painful process.

If you know a bit of CSS you could try to fix the main problem. I am unable to help you with that on such a short notice.

leftyfb, 2017-03-15:
These are the packages that were installed today. I do not know any CSS and wouldn't know where to begin to fix this myself.

2017-03-15 03:03:02 upgrade login:armhf 1:4.2-3+deb8u1 1:4.2-3+deb8u3
2017-03-15 03:03:16 upgrade libmagickwand-6.q16-2:armhf 8:6.8.9.9-5+deb8u6 8:6.8.9.9-5+deb8u8
2017-03-15 03:03:18 upgrade libmagickcore-6.q16-2:armhf 8:6.8.9.9-5+deb8u6 8:6.8.9.9-5+deb8u8
2017-03-15 03:03:32 upgrade imagemagick-common:all 8:6.8.9.9-5+deb8u6 8:6.8.9.9-5+deb8u8
2017-03-15 03:03:34 upgrade libisc-export95:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:03:34 upgrade libdns-export100:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:03:43 upgrade libisccfg-export90:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:03:44 upgrade libirs-export91:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:03:45 upgrade passwd:armhf 1:4.2-3+deb8u1 1:4.2-3+deb8u3
2017-03-15 03:04:03 upgrade bind9-host:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:04:04 upgrade libisc95:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:04:05 upgrade libdns100:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:04:06 upgrade libisccc90:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:04:07 upgrade libisccfg90:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:04:07 upgrade libbind9-90:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:04:09 upgrade liblwres90:armhf 1:9.9.5.dfsg-9+deb8u9 1:9.9.5.dfsg-9+deb8u10
2017-03-15 03:04:10 upgrade imagemagick:armhf 8:6.8.9.9-5+deb8u6 8:6.8.9.9-5+deb8u8
2017-03-15 03:04:12 upgrade imagemagick-6.q16:armhf 8:6.8.9.9-5+deb8u6 8:6.8.9.9-5+deb8u8
2017-03-15 03:04:23 upgrade libfm-data:all 1.2.3-1+rpi1 1.2.3-1+rpi2
2017-03-15 03:04:25 upgrade libfm-extra4:armhf 1.2.3-1+rpi1 1.2.3-1+rpi2
2017-03-15 03:04:25 upgrade libfm-gtk-data:all 1.2.3-1+rpi1 1.2.3-1+rpi2
2017-03-15 03:04:26 upgrade libfm-gtk4:armhf 1.2.3-1+rpi1 1.2.3-1+rpi2
2017-03-15 03:04:28 upgrade libfm4:armhf 1.2.3-1+rpi1 1.2.3-1+rpi2
2017-03-15 03:04:28 upgrade libfm-modules:armhf 1.2.3-1+rpi1 1.2.3-1+rpi2
2017-03-15 03:04:29 upgrade lxpanel:armhf 0.7.2-1+rpi17 0.7.2-1+rpi19
2017-03-15 03:04:31 upgrade lxpanel-data:all 0.7.2-1+rpi17 0.7.2-1+rpi19
2017-03-15 03:04:36 upgrade pcmanfm:armhf 1.2.3-1.1+rpi6 1.2.3-1.1+rpi7
2017-03-15 03:04:40 upgrade raspberrypi-sys-mods:all 20170127 20170313
2017-03-15 03:04:41 upgrade pi-bluetooth:armhf 0.1.2 0.1.3
2017-03-15 03:04:42 upgrade pi-greeter:armhf 0.3 0.4
2017-03-15 03:04:44 upgrade pix-plym-splash:armhf 0.7 0.9
2017-03-15 03:04:44 upgrade wiringpi:armhf 2.32 2.44
2017-03-15 03:04:46 upgrade libraspberrypi-dev:armhf 1.20161215-1 1.20170303-1
2017-03-15 03:05:04 upgrade libraspberrypi-doc:armhf 1.20161215-1 1.20170303-1
2017-03-15 03:05:33 upgrade raspberrypi-kernel:armhf 1.20161215-1 1.20170303-1
2017-03-15 03:10:09 upgrade libraspberrypi-bin:armhf 1.20161215-1 1.20170303-1
2017-03-15 03:10:10 upgrade libraspberrypi0:armhf 1.20161215-1 1.20170303-1
2017-03-15 03:10:12 upgrade raspberrypi-bootloader:armhf 1.20161215-1 1.20170303-1
2017-03-15 03:10:29 upgrade raspberrypi-ui-mods:all 1.20161206 1.20170301
2017-03-15 03:10:40 upgrade raspi-config:all 20170214 20170307

leftyfb, 2017-03-15:
I noticed the interface on my phone is also misaligned, lending more credibility to the fact that this is not related to the chromium browser on the pi but an issue with the plugin itself.
[image]

BillyBlaze, 2017-03-16:
Yes, that rules out Chromium or anything on your Pi. Anything else you have upgraded, perhaps an update in M33 Fio or OctoPrint itself? 

Nothing have changed in the controls styling in TouchUI since the release of 0.3.0 from Dec 24, 2015.

leftyfb, 2017-03-16:
The M33 plugin was updated 26 days ago. Nothing updated since that.

BillyBlaze, 2017-04-16:
The basic controls has been fixed in the latest 0.3.7 release.

closed 2017-04-16 by BillyBlaze