BillyBlaze/OctoPrint-TouchUI #113: Touch UI not staying in sync with octoprint.

nickm324, 2017-03-15:
The information displayed on the Touch UI display is not in sync with what octoprint shows in the browser.

and after a print finishes Touch UI will say for example there are 30 minutes left even though the print has finished and octoprint relects that in the browser. Then the time left counter in Touch UI will start running very fast and eventually hit the  time left  to print and stop.

BillyBlaze, 2017-03-19:
Hello,

Thanks for reporting.

In essence you're using OctoPrint; TouchUI only changes the styling. However if you could provide me some more details about the os, browser, octoprint version and the plugins you've installed then I can try to reproduce the problem.

BillyBlaze, 2017-03-27:
closed due to inactivity.

closed 2017-03-27 by BillyBlaze