BillyBlaze/OctoPrint-TouchUI #114: Scrolling not working on Adafruit 7" backpack

zarderxio, 2017-03-22:
I saw someone else with a waveshare display that had this same issue but I am not as well versed in linux to troubleshoot. I have a 7" adafruit HDMI backpack (and a waveshare 5" touchscreen) and when the scrolling it isn't registering (it just highlights text). If I do a manual install on top of the full version of rasbian and run chromium in kiosk mode, it scrolls okay. Any thoughts? Thanks!

BillyBlaze, 2017-03-27:
Hello,

Thanks for reporting, sounds like Chromium started with touch events and that is not supporting by your driver.

Could you run `cat /proc/bus/input/devices` and post output?

zarderxio, 2017-03-28:
Sure, Thanks! 

I: Bus=0003 Vendor=04d8 Product=0c02 Version=0111
N: Name="Microchip Technology Inc. AR1100 HID-MOUSE"
P: Phys=usb-3f980000.usb-1.4/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:04D8:0C02.0001/input/input0
U: Uniq=
H: Handlers=mouse0 js0 event0
B: PROP=0
B: EV=1b
B: KEY=70000 0 0 0 0 0 0 0 0
B: ABS=3
B: MSC=10

zarderxio, 2017-03-28:
So if I open chromium.xinit under ~/TouchUI-autostart and comment out all of bugfix #46 (Force Chrome to render with touch events) it seems to fix the scrolling issue. I have not tested this with the Waveshare screen I have since I will need to change some settings in config.txt to try it (possibly this weekend when I have more time). The adafruit backpack interfaces touch via usb and the waveshare is over gpio but they were both showing the same behavior so I am hopeful this will solve it for both. Not sure of any easy way for future integration but for now it sounds like I am the only one with the issue so far.

BillyBlaze, 2017-03-28:
Yeah that sounds like the culprit.

Could you verify if the below config works for you?
```
TOUCHUI_DEVICE=$(xinput --list --long | grep XITouchClass)
if [ -z "$TOUCHUI_DEVICE" ]
then
	echo "TouchUI: touch events automatically detected"
else
	echo "TouchUI: touch events enabled"
	TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
fi
```

bolsoncerrado, 2017-04-06:
OK yesterday I reported this: https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/86

Today I just saw this and tried to comment out bug46 fix, and scroll came back! :???

zarderxio, 2017-04-06:
Yea that seems to be the issue, Billyblaze posted a potential fix above your comment but I haven't had time to test. I'm going to try today. Commenting it out should work, but if a new update comes out it could break that so it's good to lock down a fix.

zarderxio, 2017-04-11:
So I just did a fresh install today but only had a waveshare gpio/hdmi screen with me and it seems to be working fine without the change. I will try to the adafruit later on.

zarderxio, 2017-04-12:
The adjust config above seems to fix it on the Adafruit backpack.

BillyBlaze, 2017-04-16:
Thank you very much for your help! I have fixed this issue. I also have added an extra variable to the default file so that user could manually force touch events if they need it.

closed 2017-04-16 by BillyBlaze

wookiemage, 2017-07-14:
Hello @BillyBlaze,
I am having the same issue with the same Adafruit backpack and 7 inch screen. I've read through all of your closed issues related to scrolling and have attempted many fixes and just can't seem to get scrolling to work. I am running Octopi .014, TouchUI 0.3.8.
No matter what I try to do, it just treats all touch events as mouse clicks.

This is what my current chromium.xinit reads:
```
#!/bin/sh

# DEFAULT VARIABLES
TOUCHUI_ARGS=""
TOUCHUI_DEBUG=""

# Uncomment the next variable to generate a full log under ~/.config/chromium/chrome_d$
# or to enable the remote deugger, forward port by haproxy or by tunnel
#TOUCHUI_DEBUG="--remote-debugging-port=9222 --enable-logging --v=100"

# Load octoprint settings for getting $PORT
[ -r /etc/default/octoprint ] && . /etc/default/octoprint

# Load touchui settings for getting $TOUCHUI_USER and $FORCE_TOUCH
[ -r /etc/default/touchui ] && . /etc/default/touchui

# Remove restore last session after crash in Chrome
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' /home/$TOUCHUI_USER/.config/c$
sed -i 's/"exit_type":"Crashed"/"exit_type":"None"/' /home/$TOUCHUI_USER/.config/chrom$
# And another one for not minified Preferences (Chrome legacy)
sed -i 's/"exited_cleanly": false/"exited_cleanly": true/' /home/$TOUCHUI_USER/.config$
sed -i 's/"exit_type": "Crashed"/"exit_type": "None"/' /home/$TOUCHUI_USER/.config/chr$

# Check if $FORCE_TOUCH is set to true
# This variable is set inside /etc/default/touchui
if [ "$FORCE_TOUCH" = true ]
then
        echo "TouchUI: touch events enabled (forced)"
        TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
else
        # Force Chrome to render with touch events
        # Bugfix #46
        TOUCHUI_DEVICE=$(xinput --list --long | grep XITouchClass)
        if [ -z "$TOUCHUI_DEVICE" ]
        then
                echo "TouchUI: touch events automatically detected"
        else
                echo "TouchUI: touch events enabled"
                TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
        fi
fi

CHROME_VERSION=$(dpkg -s chromium-browser | grep "Version: " | sed -e 's/Version: //g')
CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=10$
                --disable-java --disable-plugins --disable-extensions --disable-infoba$
                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT)'
                --start-maximized --window-position=0,0 
                $(dirname "$0")/load-screen/startup.html"

[ -r /home/$TOUCHUI_USER/TouchUI-autostart/calibration.sh ] && . /home/$TOUCHUI_USER/T$

# Start window manager, unclutter to remove cursor and start chromium as specific user$
matchbox-window-manager &
unclutter -idle 0.1 &
su $TOUCHUI_USER -c "chromium-browser $(echo $CHROME_ARGS|tr -d '\n')"
```

Here is the result of cat /proc/bus/input/devices

```
I: Bus=0003 Vendor=04d8 Product=0c02 Version=0111
N: Name="Microchip Technology Inc. AR1100 HID-MOUSE"
P: Phys=usb-3f980000.usb-1.3/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/0003:04D8:0C02.0001/input/input0
U: Uniq=
H: Handlers=mouse0 event0 js0 
B: PROP=0
B: EV=1b
B: KEY=70000 0 0 0 0 0 0 0 0
B: ABS=3
B: MSC=10

I: Bus=0003 Vendor=258a Product=0001 Version=0110
N: Name="SINO WEALTH USB KEYBOARD"
P: Phys=usb-3f980000.usb-1.4/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:258A:0001.0002/input/input1
U: Uniq=
H: Handlers=sysrq kbd leds event1 
B: PROP=0
B: EV=120013
B: KEY=10000 7 ff9f207a c14057ff febeffdf ffefffff ffffffff fffffffe
B: MSC=10
B: LED=7

I: Bus=0003 Vendor=258a Product=0001 Version=0110
N: Name="SINO WEALTH USB KEYBOARD"
P: Phys=usb-3f980000.usb-1.4/input1
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:258A:0001.0003/input/input2
U: Uniq=
H: Handlers=kbd event2 
B: PROP=0
B: EV=13
B: KEY=2a0 0 0 0 0 0 10000 2000000 3878 d801d001 1e0000 0 0 0
B: MSC=10

I: Bus=0003 Vendor=0000 Product=0538 Version=0111
N: Name=" USB OPTICAL MOUSE"
P: Phys=usb-3f980000.usb-1.5/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/0003:0000:0538.0004/input/input3
U: Uniq=
H: Handlers=mouse1 event3 
B: PROP=0
B: EV=17
B: KEY=70000 0 0 0 0 0 0 0 0
B: REL=103
B: MSC=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="ADS7846 Touchscreen"
P: Phys=spi0.1/input0
S: Sysfs=/devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input4
U: Uniq=
H: Handlers=mouse2 event4 
B: PROP=0
B: EV=b
B: KEY=400 0 0 0 0 0 0 0 0 0 0
B: ABS=1000003
```
I've tried adding the --touch-events=enabled flag to the list of chromium flags, but it doesn't seem to make any difference.

Can you please help?

Thanks,

zarderxio, 2017-07-14:
Hi wookiemage, I am not the developer but was the OP of this thread. Put a hash in front of every line of this section:

# Check if $FORCE_TOUCH is set to true
# This variable is set inside /etc/default/touchui
if [ "$FORCE_TOUCH" = true ]
then
        echo "TouchUI: touch events enabled (forced)"
        TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
else
        # Force Chrome to render with touch events
        # Bugfix #46
        TOUCHUI_DEVICE=$(xinput --list --long | grep XITouchClass)
        if [ -z "$TOUCHUI_DEVICE" ]
        then
                echo "TouchUI: touch events automatically detected"
        else
                echo "TouchUI: touch events enabled"
                TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
        fi
fi

It should look like: 
#       $CHECK=$(egrep -o "stmpe-ts" /proc/bus/input/devices)
#       if [-z "$CHECK" ]
#       then
#               echo "TouchUI: touch events automatically detected"
#               #echo "TouchUI: no stmpe-ts device"
#       else
#               echo "TouchUI: touch events enabled"
#               TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
#       fi
#else
#       echo "TouchUI: touch events enabled"
#       TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
#fi

This solved it for me.

wookiemage, 2017-07-14:
@zarderxio Thanks for the help, but I'm having a little trouble parsing your suggestion because of the markdown formatting. Any chance you could edit that so it appears in a codeblock? I can't seem to make out what lines are commented out and which are.

Thanks,

wookiemage, 2017-07-14:
@zarderxio, Thanks again! Unfortunately, that doesn't seem to have many any difference to the behavior of the screen. I modified the file, restarted the service, but it still doesn't allow me to scroll.

BillyBlaze, 2017-07-16:
Hi wookiemage,

How did you conclude that the browser is treating mouse events as touch events? Looking at this it looks like the driver is using mouse events.

Have you looked at possible other issues, such as another plugin that breaks TouchUI touch emulator?

Also could you perhaps create a new ticket? I don't get any notifications on closed issues, so I was lucky I saw this activity in the activity feed.

@zarderxio, you don't need to manual add this code anyone more, since the code you use is already inside the startup file 👍