BillyBlaze/OctoPrint-TouchUI #115: TouchUi OctoPi Pi3 Failed

PhoenixZip, 2017-03-26:
I followed this 
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

This (Basic setup and TouchUI)
https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian

And 'How to update OctoPrint on Octopi'
https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-update-the-octoprint-installation-on-my-octopi-image

So where I am now is it autolaunches TouchUI, all colorful, and fails eventually. 

I tried all the things in #44
https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/44

I will try everything in #98 next

One of the logs, tornado.access 404 GET /plugin/touchui/ping 127.0.0.1

PhoenixZip, 2017-03-26:
I have tried another fresh OctoPi image, followed the steps still have the same issues.  

Hardware used is just a 7" official touchscreen, wireless keyboard, and pi3.

PhoenixZip, 2017-03-26:
When running ~./oprint/bin/octoprint
The last line gives 
Error: [Errno 981 Address already in use]

PhoenixZip, 2017-03-26:
Okay. Ran startx and pulled up chromium and went to OctoPi.local/ octoprint loaded just fine.
And TouchUi loading screen loads on boot.
I've tried this with same results on two sd cards.

I've verified TouchUi is up to date with git-pull..
I'm just not sure why TouchUi cannot connect with octoprint if the server did launch.

BillyBlaze, 2017-03-27:
Hello,

Have you installed TouchUI within the plugin manager of OctoPrint?

PhoenixZip, 2017-03-27:
Is there a terminal line to force plugin manager to validate / install. Or is this something that has to be done via desktop startx? 
Thank you.

PhoenixZip, 2017-03-28:
So I went through startx, opened http://127.0.0.1:5000.
Got it! Plugin Manager and searched for TouchUI and all went well straight out of boot. 
Thanks again!

BillyBlaze, 2017-03-28:
Glad you figured it out & your welcome!

closed 2017-03-28 by BillyBlaze