BillyBlaze/OctoPrint-TouchUI #116: Enhancement: Reduce size of GCode View to fit screen

cyrusjk, 2017-03-26:
Id like to have the UI display the gcode layer being executed while the print is running. The Touch-UI does not scale that render down to fit in the scree; in fact, it is displayed larger than what you get with the regular Octoprint UI. Scaling it to the known window size, or at least making the size configurable would make this feature useful.

BillyBlaze, 2017-03-27:
Thanks for the suggestion. I agree that screen estate is badly optimized in that view. I will take a look at this but that will be something for in the long term.

BillyBlaze, 2017-10-18:
Thanks again for reporting, I have merged your ticket into #39 and I will close this issue for sake of cleaning up the issue tracker!

closed 2017-10-18 by BillyBlaze