BillyBlaze/OctoPrint-TouchUI #117: Controls grayed out

kdpainter, 2017-04-03:
I have TouchUI running on the latest OctoPi with a 7in display.  It all works very well except I can't delete files or initiate a print on the display.  These controls are visible but grayed out on the display.  I have to use my main computer to do these tasks.  Also, I can't cancel a print from the display.  I followed the steps on the wiki to install.  I have googled quite a lot for a solution to this issue and haven't found much.  Any ideas what might be wrong?  Regards,
kdpainter

BillyBlaze, 2017-04-03:
Hello kdpainter,

It sounds like another plugin is causing issues with OctoPrint on the Pi. Do you have any additional plugins installed?

kdpainter, 2017-04-03:
Hello,Thank you for your reply.  I believe I just have the stock plugins installed.  Here is what I have:
Announcement PluginCore WizardCuraEngineDiscoveryPlugin Manager
Software UpdateTouchUIVirtual Printer
As far as I remember, I only installed the TouchUI plugin myself.

This is a very nice interface.  When I get everything working, I know I am going to use it most of the time at the printer.

regards,kdpainter

BillyBlaze, 2017-04-03:
Thanks for the compliment and thanks for reporting this.

Okay, this definitely not caused by another plugin.

Could you try the following steps and post the output?

1. Edit this file
    ````
    nano ~/TouchUI-autostart/chromium.xinit
    ````

1. Go to line number 9 and remove the hashtag (`#TOUCHUI_DEBUG=` becomes `TOUCHUI_DEBUG=`)
1. Save and Exit (CTR-O, Enter, CTR-X)
1. Reboot TouchUI
    ````
    sudo service touchui restart
    ````
1. After the interface has fully loaded run this command:
    ````
    cat ~/.config/chromium/chrome_debug.log | grep 'CONSOLE'
    ````
1. Copy paste the output

kdpainter, 2017-04-04:
I just got back to do this.  Here is the output you requested:
 cat ~/.config/chromium/chrome_debug.log | grep 'CONSOLE'
[23753:23753:0404/011657.846334:INFO:CONSOLE(13)] "Less has finished and no sheets were loaded.", source: http://localhost:5000/static/js/lib/less.min.js (13)
[23753:23753:0404/011700.150257:INFO:CONSOLE(11891)] "Starting dependency resolution...", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (11891)
[23753:23753:0404/011700.373442:INFO:CONSOLE(632)] "TouchUI: Disabling GCodeViewer in touchscreen mode...", source: http://localhost:5000/static/webassets/packed_plugins.js?165fc06b (632)
[23753:23753:0404/011700.377306:INFO:CONSOLE(11992)] "... dependency resolution done", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (11992)
[23753:23753:0404/011700.388960:INFO:CONSOLE(12209)] "Initial application setup done, connecting to server...", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12209)
[23753:23753:0404/011700.733117:INFO:CONSOLE(10598)] "Connected to the server", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (10598)
[23753:23753:0404/011700.733391:INFO:CONSOLE(12213)] "Finalizing application startup", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12213)
[23753:23753:0404/011701.544265:INFO:CONSOLE(12117)] "Going to bind 29 view models...", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12117)
[23753:23753:0404/011701.943195:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[23753:23753:0404/011701.943361:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[23753:23753:0404/011703.300268:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[23753:23753:0404/011703.300436:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[23753:23753:0404/011703.730394:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[23753:23753:0404/011703.893494:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[23753:23753:0404/011704.178019:INFO:CONSOLE(12191)] "... binding done", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12191)
[23753:23753:0404/011704.429109:INFO:CONSOLE(12202)] "Application startup complete", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12202)

 

   - Copy paste the output

BillyBlaze, 2017-04-04:
Hmm seems there is nothing wrong with the Javascript.

Silly question but are you sure you're connected on the Pi??

kdpainter, 2017-04-04:
Yeah, I was using putty over ssh. It prints perfectly.   Everything works except some controls on touchui. Weird.

BillyBlaze, 2017-04-04:
No I mean that the web-interface of the Pi is connected. E.g. Menu icon > Connection and then within the modal its connected.

BillyBlaze, 2017-04-04:
Another silly question, but I fear this is the root cause.

Are you logged on the web-interface on the 7in display?

kdpainter, 2017-04-05:
Hello Paul,Boy, do I feel dumb.  That was the problem alright.  What threw me is I could see all the gcode files going through OctoPi on the display and I could control the printer.  I had logged in previously and thought I had checked the remember me check box so I thought I was logged in.  I would have thought if I was not logged in, the display wouldn't show anything useful.  Sorry for wasting your time.  One question, is there a way to disable the log in requirement on TouchUI only?  This device will live inside my house and I am behind a NATed firewall.  The login not really needed in my situation.

Thank you so much!  This is awesome! 

I am just coming back to the RPi and OctoPrint.  I had tried it a long time ago and was having crashes on a RPi B+ (where it would stop and leave the heaters on).  I am not sure what that was about but I quit using it out of safety concerns.  But now I have a RPi 3b.  No such problems and that was before your touch UI.  Wow! Thanks so much.  You need a contribute button :)Regards,kdpainter

BillyBlaze, 2017-04-05:
Awesome, glad you figured out the problem. And no problem, happens to the best!

I was made aware by @foosel that you could enable autologinLocal within the config of OctoPrint. See how you can do this on the [docs of octoprint](http://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control)

closed 2017-04-05 by BillyBlaze