BillyBlaze/OctoPrint-TouchUI #119: Doesn´t leave "Connecting to TouchUI"

MaxMihawk, 2017-04-09:
I have a very big miracle problem. My RPi3 with OctoPi was connected to a 7" touchscreen. The touchscreen only displays the "Connecting to TouchUI" screen.

[image]

But.... my laptop is working with the interface **at the same time**, without any problems.

[image]

How could this happen?!?!?

config.txt part for the screen:
```
# 7 Inch HDMI Capacitive IPS Display LCD for Raspberry Pi 1024 x 600
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```
Screen: http://www.waveshare.com/wiki/7inch_HDMI_LCD_(C)

BillyBlaze, 2017-04-12:
Hello,

Thanks for reporting. This is a big miracle sofar... Couple of questions;

Does the message "Connecting to TouchUI" show a number at the front?
Could you share your octoprint.log?

MaxMihawk, 2017-04-16:
Hi BillyBlaze,

their is no number as you can see on the photo. The log says:

```
2017-04-14 06:04:23,704 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2017-04-14 06:04:23,705 - octoprint.server - INFO - OctoPrint 1.3.2 (master branch)
2017-04-14 06:04:23,706 - octoprint.plugin.core - INFO - 8 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/softwareupdate
|  TouchUI (0.3.6) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/virtual_printer
2017-04-14 06:04:23,706 - octoprint.server - INFO - ------------------------------------------------------------------------------
2017-04-14 06:04:23,700 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 179.28ms
2017-04-14 06:04:26,717 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.46ms
2017-04-14 06:04:29,736 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.85ms
2017-04-14 06:04:32,752 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.31ms
2017-04-14 06:04:35,768 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.80ms
2017-04-14 06:04:38,785 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.66ms
2017-04-14 06:04:41,803 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 9.05ms
2017-04-14 06:04:44,821 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 9.60ms
2017-04-14 06:04:47,834 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 8.35ms
2017-04-14 06:04:50,850 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 8.22ms
2017-04-14 06:04:53,866 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.91ms
2017-04-14 06:04:56,882 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 9.18ms
2017-04-14 06:04:59,899 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.85ms
2017-04-14 06:05:02,915 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.89ms
2017-04-14 06:05:05,933 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 9.58ms
2017-04-14 06:05:08,947 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 8.41ms
2017-04-14 06:05:11,965 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 9.54ms
2017-04-14 06:05:14,983 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 9.18ms
2017-04-14 06:05:17,999 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 6.80ms
2017-04-14 06:05:21,017 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.91ms
2017-04-14 06:05:24,033 - tornado.access - WARNING - 404 GET /plugin/touchui/echo (127.0.0.1) 7.87ms
```

BillyBlaze, 2017-04-16:
Thanks for the log!

Is there perhaps a file named `plugin_touchui.log` within the same log folder?

MaxMihawk, 2017-04-16:
Hi. Their is no such file. I can give you the "plugin_softwareupdate_console.log".
```
2017-01-07 17:35:51,984   /home/pi/oprint/bin/python -m pip install https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/0.3.5.zip
2017-01-07 17:35:55,278 > Collecting https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/0.3.5.zip
2017-01-07 17:35:58,746 > Downloading https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/0.3.5.zip (753kB)
2017-01-07 17:36:03,771 > Requirement already satisfied (use --upgrade to upgrade): OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg (from TouchUI==0.3.5)
2017-01-07 17:36:04,075 > Requirement already satisfied (use --upgrade to upgrade): flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,076 > Requirement already satisfied (use --upgrade to upgrade): werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,076 > Requirement already satisfied (use --upgrade to upgrade): tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,077 > Requirement already satisfied (use --upgrade to upgrade): sockjs-tornado<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,077 > Requirement already satisfied (use --upgrade to upgrade): PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,077 > Requirement already satisfied (use --upgrade to upgrade): Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,078 > Requirement already satisfied (use --upgrade to upgrade): Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,078 > Requirement already satisfied (use --upgrade to upgrade): Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,078 > Requirement already satisfied (use --upgrade to upgrade): Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,079 > Requirement already satisfied (use --upgrade to upgrade): markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.6-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,079 > Requirement already satisfied (use --upgrade to upgrade): pyserial<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,080 > Requirement already satisfied (use --upgrade to upgrade): netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.17-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,080 > Requirement already satisfied (use --upgrade to upgrade): watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,081 > Requirement already satisfied (use --upgrade to upgrade): sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,081 > Requirement already satisfied (use --upgrade to upgrade): netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.0-py2.7-linux-armv7l.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,082 > Requirement already satisfied (use --upgrade to upgrade): pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,083 > Requirement already satisfied (use --upgrade to upgrade): rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,083 > Requirement already satisfied (use --upgrade to upgrade): pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,084 > Requirement already satisfied (use --upgrade to upgrade): requests<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,084 > Requirement already satisfied (use --upgrade to upgrade): semantic-version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,085 > Requirement already satisfied (use --upgrade to upgrade): psutil<3.3,>=3.2.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-3.2.1-py2.7-linux-armv7l.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,085 > Requirement already satisfied (use --upgrade to upgrade): Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->TouchUI==0.3.5)
2017-01-07 17:36:04,085 > Requirement already satisfied (use --upgrade to upgrade): awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint-> [... cut ...]
2017-01-07 17:36:20,100 ! You are using pip version 8.1.1, however version 9.0.1 is available.
2017-01-07 17:36:20,122 ! You should consider upgrading via the 'pip install --upgrade pip' command.
2017-02-24 19:11:38,039   /home/pi/oprint/bin/python "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py" --branch= --force=false "/home/pi/OctoPrint" 1.3.1
2017-02-24 19:11:46,568 ! From https://github.com/foosel/OctoPrint
...
```

BillyBlaze, 2017-04-16:
Hmm, okay, let's dive a bit deeper.

Could you try the following steps and post the output?

1. Edit this file
    ````
    nano ~/TouchUI-autostart/chromium.xinit
    ````

1. Go to line number 9 and remove the hashtag (`#TOUCHUI_DEBUG=` becomes `TOUCHUI_DEBUG=`)
1. Save and Exit (CTR-O, Enter, CTR-X)
1. Reboot TouchUI
    ````
    sudo service touchui restart
    ````
1. After the interface has fully loaded run this command:
    ````
    cat ~/.config/chromium/chrome_debug.log | grep 'CONSOLE'
    ````
1. Copy paste the output

MaxMihawk, 2017-04-16:
Actually i dont have the printer at my home, so this will need some more weeks :) . I´ll get to you soon.

hegmar, 2017-05-23:
got the same problem (same screen), but i ve no sign (right corner = less power) like him 
and i ve counting numbers and after a failed of the TouchUI
tried to edit this file nano ~/TouchUI-autostart/chromium.xinit but this folder is empty?

hegmar, 2017-05-23:
now created that file (copy of https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/chromium.xinit) and now there is a white screen with... your file was not found :/

BillyBlaze, 2017-05-24:
Hello, I have no idea why the required folder is empty. Did you follow somebody else his tutorial? It sounds like after you created the file, it still has missing files. 

After deleting the whole folder `~/TouchUI-autostart/` follow the instructions from step 7 in this guide: https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

hegmar, 2017-05-24:
yes i followed that tut... everything was workin except teststart of the chromium-browser :/

i@octopi:~ $ uname -a
Linux octopi 4.9.29-v7+ #1000 SMP Sun May 21 20:11:31 BST 2017 armv7l GNU/Linux
pi@octopi:~ $ sudo vcgencmd version
May 15 2017 17:01:07
Copyright (c) 2012 Broadcom
version 9469ea3706e34c4de62f38a5008f69a429b4b43e (clean) (release)
pi@octopi:~ $ apt-cache policy chromium-browser
chromium-browser:
  Installed: 56.0.2924.84-0ubuntu0.14.04.1.1000
  Candidate: 56.0.2924.84-0ubuntu0.14.04.1.1000
  Version table:
 *** 56.0.2924.84-0ubuntu0.14.04.1.1000 0
        500 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
pi@octopi:~ $ cat /proc/device-tree/model
Raspberry Pi 3 Model B Rev 1.2pi@octopi:~ $ Installed: 2:1.18.4-0ubuntu0.2
-bash: Installed:: command not found
pi@octopi:~ $ chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0523/163311.933506:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[1127:1127:0523/163311.940035:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[1127:1127:0523/163311.949043:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display:

BillyBlaze, 2017-05-24:
The teststart of chromium looks fine.

You where saying that you get a white screen with "Your file was not found" after manually creating the file? Do you even have the folder `~/TouchUI-autostart/load-screen/` with 3 files in it? If not, do you still have enough storage space left on your SD?

hegmar, 2017-05-24:
hmmm i ve to look @ home... but i ve a 16gb card and i expand filesystem after first boot...

hegmar, 2017-05-24:
~/TouchUI-autostart was empty, i copied the files from the internet to the raspberry over filezilla and now it works fine :)
thx 4 helping BillyBlaze 
[image]

BillyBlaze, 2017-06-03:
@MaxMihawk any status update?

MaxMihawk, 2017-06-03:
Hi. My dad want a delta for himself, so i bought a second pi and at the time of the setup I'll test the screen on the fresh installation. I'll post my result after.
Actually I'm on holiday so it will need some more weeks :).
Cheers, MaxMihawk

Jimbles, 2017-06-05:
hey! I thought I would just mention that in [the tutorial](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup: -Boot-to-Browser-(OctoPi-or-Jessie-Light)  you say to use `Option "fbdev" "/dev/fb0"` only for the official RPi Touchscreen, and `fb1` for others. But my totally non offical Elecrow 5inch screen was being mounted as `fb0` too. Which cost me about half an hour of my life!

Also giving the full list of "ok errors" might be useful:
```
/usr/bin/chromium-browser: line 138: lsb_release: command not found
/usr/bin/chromium-browser: line 139: lsb_release: command not found
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0605/122948.230046:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[3904:3904:0605/122948.236249:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[3904:3904:0605/122948.244840:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display:
```

Thanks!

BillyBlaze, 2017-06-05:
Hey,

Please feel free to edit the Wiki and to update it with your findings. 👍

Jimbles, 2017-06-05:
Oh! I didn't realise that I could! Sorry for that, I guess that seemed passive aggressive which was not my intention 👍

MaxMihawk, 2017-07-04:
Hi. You can close the issue. After a complete new install on another pi3, all is working. I´ll install the same configuration on the "old" pi3 too 👍 .

BillyBlaze, 2017-07-16:
Hi, okay, glad it works, I was very curious why, perhaps the files where corrupted?

Well anyhow, let's close this.

closed 2017-07-16 by BillyBlaze