BillyBlaze/OctoPrint-TouchUI #120: [ Error writing /etc/X11/xorg.conf.d/99-fbdev.conf: No such file or directory ]

Shahriar57, 2017-04-20:
Hello
I tried many time this setup step by step 
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

when i run Run sudo nano /etc/X11/xorg.conf.d/99-fbdev.conf
and add 
Section "Device"  
  Identifier "touchscreen"
  Driver "fbdev"
  Option "fbdev" "/dev/fb1"
EndSection
then CRTL+O , Enter 
get 
 [ Error writing /etc/X11/xorg.conf.d/99-fbdev.conf: No such file or directory 

 i search in Issue and find this issue 
https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/44
then Run sudo apt-get install lxde-core
and after finish reboot 
but again i got same error 
what shall i do ?

BillyBlaze, 2017-04-20:
Hello, The instructions talk about ```sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf``` and not ```/etc/X11/xorg.conf.d/99-fbdev.conf```

BillyBlaze, 2017-04-24:
Have you had any luck yet?

BillyBlaze, 2017-04-28:
Closed due to inactivity. Please feel free to resume conversation if needed.

closed 2017-04-28 by BillyBlaze