BillyBlaze/OctoPrint-TouchUI #121: Mobile view on Mac Safari instead of classic view

kr15uk, 2017-04-28:
When opening Octoprint Web on Safari 10.1 +  macOS 10.12.4 it loads TouchUI despite it's a desktop machine. Works fine on macOS + Chrome.
Tried to do "User agent" as Chrome in Safari's Develop menu but still loads Touch UI.
Don't get me wrong, I do like it but I do prefer classic look on desktops and TouchUI on mobile devices.

Thanks.

BillyBlaze, 2017-04-28:
Hello, thanks for reporting, I am afraid that the browser is reporting itself as a touch device and therefor TouchUI gets activated. I am afraid that ```'ontouchstart' in window```returns true.  Which has been fixed in Chrome but not in Safari.

However there are some workarounds:
1. You can tell TouchUI to disable itself in a specific browser. It will no longer activate itself (once an user disabled or enabled it will respect that choice through localstorage or a cookie) see '[TouchUI settings](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Configuration#user-settings)'.
1. You can disable the auto-detection and manually enable TouchUI if needed. See '[Settings > TouchUI](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Configuration#global-settings)'

I could try to use the new feature detection from [Modernizr](https://github.com/Modernizr/Modernizr/blob/5eea7e2a213edc9e83a47b6414d0250468d83471/feature-detects/touchevents.js) but I am afraid that this will on the long term give the same issues.

BillyBlaze, 2017-05-05:
Any feedback before I close this issue?

closed 2017-05-08 by BillyBlaze