BillyBlaze/OctoPrint-TouchUI #122: Chromium does not start after update

Petclaud, 2017-04-30:
Hi,

I have done some updates on my Raspberry and now Chromium does not start any more. The problem looks to me very similar as described here: 
https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/104 
and here:
https://github.com/RPi-Distro/repo/issues/58

I have a 7" display though HDMI connected to the Raspberry Pi3. Before the update, when I booted the system, the TouchUI would be started and displayed. Now the system boots, but the 7" display shows just text and and in the end turns black. The Touch-UI does not start automatically.

When I type "chromium-browser" via Putty, I get the following error:
 
pi@octopi:~ $ chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0430/064649.197581:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[915:915:0430/064649.208531:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[915:915:0430/064649.225564:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display:

Any thoughts/tips/advise...? Thx in advance!

BillyBlaze, 2017-04-30:
Hello,

Thanks for reporting.
First off have you tried to fix this with the instructions from #81

Petclaud, 2017-04-30:
Hi thx for the very fast response! Yes I have:

pi@octopi:~ $ sudo service touchui stop
pi@octopi:~ $ cd ~/TouchUI-autostart/ && git pull
Already up-to-date.
pi@octopi:~/TouchUI-autostart $ sudo cp ~/TouchUI-autostart/touchui.init /etc/in                                                            it.d/touchui
pi@octopi:~/TouchUI-autostart $ sudo chmod +x /etc/init.d/touchui
pi@octopi:~/TouchUI-autostart $ sudo systemctl daemon-reload
pi@octopi:~/TouchUI-autostart $ sudo service touchui start
pi@octopi:~/TouchUI-autostart $

also after a reboot, yhe display, displays: octopi login:_

Petclaud, 2017-04-30:
Just to add: I not only ran the update commands, I also performed a calibration of the screen. Maybe this is also in play here. I did te following:
cd /boot
cd LCD-show
sudodpkg -i -B xinput-calibrator_0.7.5-1_armhf.deb
DISPLAY=:0.0 xinput_calibrator
and modified the parameters via: sudonano /etc/X11/xorg.conf.d/99-calibration.conf

.... maybe this has something to do with my issue

BillyBlaze, 2017-04-30:
Your welcome, I just did an update myself and it kept working for me.

So there should be another issue.

Could you run;
```sudo service touchui stop```

And then manually run and Copy-Paste the output to here:
```sudo xinit ~/TouchUI-autostart/chromium.xinit```

Petclaud, 2017-04-30:
Wow super fast support. Great, thx a bunch. I got the following:
pi@octopi:~ $ sudo service touchui stop
pi@octopi:~ $ sudo xinit ~/TouchUI-autostart/chromium.xinit

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.9.25-v7+ #994 SMP Fri Apr 28 16:56:00 BST 2017 armv7l
Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1024 bcm2708_fb.fbheight=600 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.l                  pm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sun Apr 30 07:39:59 2017
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
Parse error on line 1 of section InputClass in file /usr/share/X11/xorg.conf.d/99-fbdev.conf
        "Section" is not a valid keyword in this section.
(EE)
Fatal server error:
(EE) no screens found(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
pi@octopi:~ $

_______________

BillyBlaze, 2017-04-30:
Hmm seems like something is wrong with this config file: ```Parse error on line 1 of section InputClass in file /usr/share/X11/xorg.conf.d/99-fbdev.conf```

Could you run ```cat /usr/share/X11/xorg.conf.d/99-fbdev.conf``` and post the config?

BillyBlaze, 2017-04-30:
also another thing to look at, the calibration file you created ```/etc/X11/xorg.conf.d/99-calibration.conf```

could it be that its missing the EndSection line.

Petclaud, 2017-04-30:
Hi thx!. Pls find herewith the result of the first advice:

Section "Device"
  Identifier "touchscreen"
  Driver "fbdev"
  Option "fbdev" "/dev/fb0"
EndSection

BillyBlaze, 2017-04-30:
That looks ok, but I am pretty sure that another config file is a missing an EndSection line:

````
section InputClass in file /usr/share/X11/xorg.conf.d/99-fbdev.conf
"Section" is not a valid keyword in this section.
````

I say this because in the file 99-fbdev.conf there is no `section InputClass` defined. So it tries to now load a section within a section.

The concatenated config file (done by X11) now looks something like:

````
Section InputClass
Someproperties "true"

Section "Device"
Identifier "touchscreen"
Driver "fbdev"
Option "fbdev" "/dev/fb0"
EndSection
````

while it should look like:

````
Section InputClass
Someproperties "true"
EndSection

Section "Device"
Identifier "touchscreen"
Driver "fbdev"
Option "fbdev" "/dev/fb0"
EndSection
````

Petclaud, 2017-04-30:
Yes thx Paul, you are right; the EndSection statement was indeed missing...

and IT WORKS AGAIN!!! Hurray. 

Paul de Vries reads very Dutch, so; hartelijk dank voor deze fantastische ondersteuning. Een voorbeeld voor vele ontwikkelaars. Dank!

BillyBlaze, 2017-04-30:
Awesome, glad it's working again :)

Your assumption is right, so; Dankjewel voor het compliment en graag gedaan!

closed 2017-04-30 by BillyBlaze

20161220-rootfest, 2017-05-04:
Hi, I use a 3.5" display with this driver http://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)

I already tryed 

--------------------------------------------

Section "Device"
  Identifier "touchscreen"
  Driver "fbdev"
  Option "fbdev" "/dev/fb0"
EndSection

--------------------------------------------

Section "Device"
  Identifier "touchscreen"
  Driver "fbdev"
  Option "fbdev" "/dev/fb1"
EndSection

--------------------------------------------

Section "Device"
        Identifier      "Allwinner A10/A13 FBDEV"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb1"
        Option          "SwapbuffersWait" "true"
EndSection

--------------------------------------------

but nothing works display is only white

root@octopi:/# chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0504/134323.232554:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[2365:2365:0504/134323.239364:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[2365:2365:0504/134323.247852:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display:

root@octopi:/# sudo xinit ~/TouchUI-autostart/chromium.xinit

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.9.25-v7+ #994 SMP Fri Apr 28 16:56:00 BST 2017 armv7l
Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu May  4 13:37:05 2017
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found.
TouchUI: touch events automatically detected
wm_set_cursor_visibility: visible using XCreateFontCursor
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0504/133718.658086:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
^Cxinit: connection to X server lost

waiting for X server to shut down [2121:2121:0504/133758.066334:ERROR:chrome_browser_main_extra_parts_x11.cc(62)] X IO error received (X server probably went away)
(II) Server terminated successfully (0). Closing log file.

xinit: unexpected signal 2

BillyBlaze, 2017-05-04:
@20161220-rootfest a white screen indicates there is a problem with your driver. Are you sure you have selected the right driver? If your Pi boots you should briefly see a white screen and then it should show a command line interface during the initial boot.

20161220-rootfest, 2017-05-05:
Thanks.
I reinstalled everything on a new sd card....(Octopi , LCD Driver and TouchUI)
The driver is right I just had to downgrade the kernel to 4.4.50.
Now the screen is working but not the TouchUI on the screen.

----------------------------------------------------------------------------------------------

root@octopi:/home/pi# chromium-browser
/usr/bin/chromium-browser: line 138: lsb_release: command not found
/usr/bin/chromium-browser: line 139: lsb_release: command not found
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0505/161615.327875:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[829:829:0505/161615.399560:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[829:829:0505/161615.492291:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display:

----------------------------------------------------------------------------------------------

root@octopi:/home/pi# xinit ~/TouchUI-autostart/chromium.xinit

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l
Kernel command line: 8250.nr_uarts=1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa02082 bcm2709.serial=0xf4d27df4 smsc95xx.macaddr=B8:27:EB:D2:7D:F4 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Fri May  5 16:17:13 2017
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) no screens found(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

BillyBlaze, 2017-05-05:
You're almost there. The first part looks good but the second part is giving the error 'no screens found'.

What config did you paste into `/usr/share/X11/xorg.conf.d/99-fbdev.conf`?

20161220-rootfest, 2017-05-05:
I tryed

Section "Device"
Identifier "touchscreen"
Driver "fbdev"
Option "fbdev" "/dev/fb0"
EndSection

Section "Device"
Identifier "touchscreen"
Driver "fbdev"
Option "fbdev" "/dev/fb1"
EndSection

Section "Device"
Identifier "Allwinner A10/A13 FBDEV"
Driver "fbturbo"
Option "fbdev" "/dev/fb1"
Option "SwapbuffersWait" "true"
EndSection

Section "Device"
Identifier "Allwinner A10/A13 FBDEV"
Driver "fbturbo"
Option "fbdev" "/dev/fb0"
Option "SwapbuffersWait" "true"
EndSection

BillyBlaze, 2017-05-05:
Ok, try only the following config:

```
Section "Device"
  Identifier "touchscreen"
  Driver "fbdev"
  Option "fbdev" "/dev/fb1"
EndSection
```

and run `xinit ~/TouchUI-autostart/chromium.xinit` as root again to see the output. The output should be familiar like the log you posted yesterday (which looked ok).

20161220-rootfest, 2017-05-05:
I dont know why but now it works !!! 

but the camera isnt working.

---------------------------------------------------------------------------------------------------

root@octopi:/home/pi# xinit ~/TouchUI-autostart/chromium.xinit

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l
Kernel command line: 8250.nr_uarts=1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa02082 bcm2709.serial=0xf4d27df4 smsc95xx.macaddr=B8:27:EB:D2:7D:F4 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Fri May  5 16:40:00 2017
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
sed: can't read /home/pi/.config/chromium/Default/Preferences: No such file or directory
sed: can't read /home/pi/.config/chromium/Default/Preferences: No such file or directory
sed: can't read /home/pi/.config/chromium/Default/Preferences: No such file or directory
sed: can't read /home/pi/.config/chromium/Default/Preferences: No such file or directory
TouchUI: touch events automatically detected
/usr/bin/chromium-browser: line 138: lsb_release: command not found
/usr/bin/chromium-browser: line 139: lsb_release: command not found
wm_set_cursor_visibility: visible using XCreateFontCursor
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0505/164003.840841:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[936:936:0505/164007.243541:ERROR:sandbox_linux.cc(343)] InitializeSandbox() called with multiple threads in process gpu-process.

(chromium-browser:872): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
/usr/bin/dbus-launch terminated abnormally without any error message
^Cxinit: connection to X server lost

waiting for X server to shut down [936:936:0505/164117.632462:ERROR:x11_util.cc(87)] X IO error received (X server probably went away)
[872:872:0505/164117.650340:ERROR:chrome_browser_main_extra_parts_x11.cc(62)] X IO error received (X server probably went away)
(II) Server terminated successfully (0). Closing log file.

xinit: unexpected signal 2

maxoutny, 2017-05-20:
After new install and trying to run: chromium-browser
have errors below:
pi@octopi:~ $ chromium-browser
/usr/bin/chromium-browser: line 138: lsb_release: command not found
/usr/bin/chromium-browser: line 139: lsb_release: command not found
/usr/lib/chromium-browser/chromium-browser: symbol lookup error: /usr/lib/chromium-browser/chromium-browser: undefined symbol: mmal_vc_init_fd

any help would be appreciated. I am using Raspberry Pi 3 with 7" touch screen and fresh install of OctoPi

Braxstonwithans, 2018-01-08:
I'm having the same issue. My chromium browser only worked for a split second after the initial installation. Now the browser will not open at all. The icon is there and when I click on it I get a loading cursor.

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=9b19e749-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support) 
Current version of pixman: 0.33.3
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Mon Jan  8 20:54:26 2018
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.

BillyBlaze, 2018-01-08:
Locked due to multiple necroposting, please feel free to create your own issue.