BillyBlaze/OctoPrint-TouchUI #123: Webcam shows broken image, only on Pi

lewisd32, 2017-05-08:
Hello,

First off, just want to say: Great bit of software. I'm really looking forward to using this! It's exactly what I was looking for.

That said... I'm having a wee spot of trouble with the webcam integration. I'm running TouchUI on Octopi 1.3.2 on a Pi 3 with official 7" touchscreen, and a USB webcam. When I connect to Octoprint from my desktop, using the normal Octoprint UI *or* TouchUI, the webcam works flawlessly. It works on my iPhone too. However, from the touchscreen on the Pi, I just get the "broken image" icon.

The diagnostic steps about Snapstream don't seem particularly relevant to this problem, but I tried them anyway. http://pi-ip:8080/?action=snapshot gives me a still image just fine.  I was going to try to figure out what's up with the broken image on the Pi using the chrome devtools, but I can't seem to get them to open, even with a mouse and keyboard attached to the Pi.

Not sure what to do to diagnose this. Any ideas?

lewisd32, 2017-05-08:
I just found this in `octoprint.log` when I go to the webcam tab on TouchUI on the Pi:
```2017-05-08 02:49:43,647 - tornado.access - WARNING - 404 GET /webcam/?action=stream&1494211783609 (127.0.0.1) 23.30ms```
I already changed the snapstream config to `/?action=snapshot` according to the instructions, since that's the URL that I was getting an image on. Not sure what's going on.

kelexel, 2017-05-10:
Same issue here..
Webcam show up on regular browser, but when using a pitft hooked to the pi, the webcam preview is broken :(

BillyBlaze, 2017-05-13:
Hmm, it looks like the browser is connection to the OctoPrint server and not the webcam server. Hence the access warning ends up in the OctoPrint log.

Have your tried setting the prefix of webcam url to: `http://octopi.local:8080/`?

We need an absolute url and not a relative one, this is becuase your touchscreen runs on `http://localhost:5000/` and we need to overwrite the port to 8080.

lewisd32, 2017-05-13:
I can give that a try when I'm home again, but I'm a bit puzzled. If that's the problem, I'm not sure why it would work when connecting to the touchUi from my desktop and phone. It's only from the Pi that it doesn't work.

lewisd32, 2017-05-16:
Oh, I see. The touchscreen on the Pi connects to a different port.

If I put "http://10.0.0.15:8080/?action=snapshot" in the snapstream config, it works on the desktop and the touchscreen now. I left "Only use snapstream when mjpg-streamer fails" checked, so on the desktop and phone I get smooth streaming, and on the Pi I get the 4 FPS of snapstream.

closed 2017-05-16 by lewisd32