BillyBlaze/OctoPrint-TouchUI #126: TouchUI interface jumpy

xsimonsaysx, 2017-05-20:
First let me state I'm completely new at this so it's taken me a week to get TouchUI setup and autostarting at all.  Better stated I laid down my Linux aspirations over 15 years ago and am relearning everything again.

Here's my issue:  When I try to use the touch interface of my official rpi 7" screen with TouchUI it works until I try to scroll.  At that point TouchUI seems to get stuck and once I get to the bottom of my screen and try to start scrolling from the top again it jumps back to the last pointer location essentially making it unusable.  Outside of TouchUI everything works fine.  I built my setup from a full Jessie install as I could never get an octopi setup to work at all.

I have been trying to retrace my steps through all of the setup steps to see what/if I may have missed anything and have had little luck so far.  
1 error I have come across is when I try to do this step:
sudo sed -i 's/allowed_users=console/allowed_users=anybody/' /etc/X11/Xwrapper.config
I get this in response:
sed: can't read /etc/X11/Xwrapper.config: No such file or directory
and indeed there is no such file in the X11 directory.
I don't know if this is the source of my issue or how to fix it and any would be appreciated.

BillyBlaze, 2017-05-21:
Hello, 

You can ignore the error about ```sudo sed [...]```. (That is legacy code in your case.)

It sounds like a third party plugin is causing some issues, do you have any third party plugins installed?

xsimonsaysx, 2017-05-21:
I followed the steps in foosels guide here:
https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian
I skipped the part about setting up iptables for mjpg streamer for now and I stopped at the about setting up HAproxy so every step through System commands portion was completed.  From there I switched to your guide for TouchUI and went through each step.  The only other plugins I've installed in Octoprint are Automatic Shutdown and STL viewer.  The final thing I did was follow your guide on screen calibration thinking that was my problem.  After that I started checking my installs and going through my files and directories piece by piece before contacting you.  Hope this helps.

BillyBlaze, 2017-05-21:
Thanks for the information.

Just to be safe. You followed my guide that is located on this url: https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(Jessie-Full) ?

Could you also try the following steps and post the output?

1. Edit this file
    ````
    nano ~/TouchUI-autostart/chromium.xinit
    ````

1. Go to line number 9 and remove the hashtag (`#TOUCHUI_DEBUG=` becomes `TOUCHUI_DEBUG=`)
1. Save and Exit (CTR-O, Enter, CTR-X)
1. Reboot TouchUI
    ````
    sudo service touchui restart
    ````
1. After the interface has fully loaded try scrolling and reproduce the problem
1. Run this command:
    ````
    cat ~/.config/chromium/chrome_debug.log | grep 'CONSOLE'
    ````
1. Copy paste the output

xsimonsaysx, 2017-05-21:
Yes, that is the guide I followed.  I will try to get the debug info for you once I get home from work this evening.  In the meantime another thought occurred to me.  The problem I'm having reminds me of a click and drag operation with a mouse.  It's like it's getting stuck in that mode.

BillyBlaze, 2017-05-21:
Hmm, now that you say that, there has been some changes to the drag and drop functionality within the 1.3.3 pre-release. Are you running OctoPrint 1.3.3 or are you still on 1.3.2?

xsimonsaysx, 2017-05-21:
Still running 1.3.2.....made sure I stayed with stable releases on everything at least until I feel completely reacquainted with what I'm doing.

xsimonsaysx, 2017-05-21:
[1193:1193:0521/093545.949821:INFO:CONSOLE(13)] "Less has finished and no sheets were loaded.", source: http://localhost:5000/static/js/lib/less.min.js (13)
[1193:1193:0521/093548.789125:INFO:CONSOLE(11891)] "Starting dependency resolution...", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (11891)
[1193:1193:0521/093549.054931:INFO:CONSOLE(7793)] "TouchUI: Disabling GCodeViewer in touchscreen mode...", source: http://localhost:5000/static/webassets/packed_plugins.js?e6dc2f64 (7793)
[1193:1193:0521/093549.060509:INFO:CONSOLE(11992)] "... dependency resolution done", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (11992)
[1193:1193:0521/093549.071578:INFO:CONSOLE(12209)] "Initial application setup done, connecting to server...", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12209)
[1193:1193:0521/093549.460340:INFO:CONSOLE(10598)] "Connected to the server", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (10598)
[1193:1193:0521/093549.465205:INFO:CONSOLE(12213)] "Finalizing application startup", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12213)
[1193:1193:0521/093549.984219:INFO:CONSOLE(12117)] "Going to bind 31 view models...", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12117)
[1193:1193:0521/093550.413788:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[1193:1193:0521/093550.414124:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[1193:1193:0521/093551.279383:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[1193:1193:0521/093551.279570:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[1193:1193:0521/093551.631307:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[1193:1193:0521/093551.839563:INFO:CONSOLE(12158)] "Did not bind view model", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12158)
[1193:1193:0521/093552.340859:INFO:CONSOLE(12191)] "... binding done", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12191)
[1193:1193:0521/093552.499183:INFO:CONSOLE(12202)] "Application startup complete", source: http://localhost:5000/static/webassets/packed_core.js?1dfc6ea5 (12202)

xsimonsaysx, 2017-05-21:
That's what I got after doing the steps you requested.

xsimonsaysx, 2017-05-21:
Did another reboot and ran it again, this time the log started with:

[814:814:0521/094743.102442:INFO:CONSOLE(0)] "XMLHttpRequest cannot load http://localhost:5000/plugin/touchui/ping. No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'null' is therefore not allowed access. The response had HTTP status code 404.", source: file:///home/pi/TouchUI-autostart/load-screen/startup.html (0)

after that the rest looks the same as the previous post

BillyBlaze, 2017-05-21:
Hmm, everything looks absolutely fine... and I just can't reproduce any strange behavior on my own official rpi 7".

Could you perhaps make a video how you can reproduce this issue?

xsimonsaysx, 2017-05-21:
[Sample-1.zip](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/1017112/Sample-1.zip)

Had to jump through a few hoops to get this, but here it is.  I currently have the plastic shipping film on the screen, but have tried it with and without that in place and had the same results.

BillyBlaze, 2017-05-21:
Hmm, that looks like it's in mouse-mode while the browser is sending touch events.

Can you try this:
1. Edit the default file:
    ```sudo nano /etc/default/touchui```

1. Go all the way down and change ```FORCE_TOUCH=false``` into ```FORCE_TOUCH=true```
1. Save and Exit (CTR-O, Enter, CTR-X)
1. Restart touchui ```sudo service touchui restart```

xsimonsaysx, 2017-05-21:
That completely fixed it.  Thank you, I may never have found that setting.  Now I just need to figure out how to get my WLAN setup and probably HAproxy so I can remote in.  Again thank you for your help.

closed 2017-05-21 by xsimonsaysx