BillyBlaze/OctoPrint-TouchUI #127: erros to start chromium 138, 139

maxoutny, 2017-05-20:
After new install and trying to run: chromium-browser
have errors below:
pi@octopi:~ $ chromium-browser
/usr/bin/chromium-browser: line 138: lsb_release: command not found
/usr/bin/chromium-browser: line 139: lsb_release: command not found
/usr/lib/chromium-browser/chromium-browser: symbol lookup error: /usr/lib/chromium-browser/chromium-browser: undefined symbol: mmal_vc_init_fd

any help would be appreciated. I am using Raspberry Pi 3 with 7" touch screen and fresh install of OctoPi

BillyBlaze, 2017-05-21:
Hello, As instructed in Step 9 of the [WIki](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)) you need to upgrade your kernel.

maxoutny, 2017-05-21:
Thank you for quick reply. I fixed that problem. It was user mistake during installation.
When I add lines in configuration file for display I added config for HDMI connection not
for 7" attached by ribbon cable. After mistake corrected all works as expected.
Again, thank you for all your hard work.

closed 2017-05-21 by maxoutny