BillyBlaze/OctoPrint-TouchUI #130: Change the text on bootup

Mte90, 2017-06-22:
Actually when the TouchUi start the logo is TouchUI text, it is possible to change in the settings with someone else?

BillyBlaze, 2017-06-23:
There is no way to change this within the settings. The TouchUI logo are inline SVG, the only way you can change this is by manual editing [the html](https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/load-screen/startup.html) within the bootloader found under `~/TouchUI-autostart/load-screen/startup.html`

Mte90, 2017-06-23:
A file upload for a SVG can be an idea for it :-)

BillyBlaze, 2017-10-17:
That is sadly not possible, the bootloader are static files that are loaded directly from the file system, There is no server, database or OctoPrint involved, which makes customization hard and therefor it's better for those few cases to manually change the file themself.

Never the less thanks for your feedback!

closed 2017-10-17 by BillyBlaze

Mte90, 2017-10-18:
We are talking about the touchui logo, it is not possible to do a symlink to a different file?

BillyBlaze, 2017-10-18:
Yes, but you would still be required to edit the static HTML and refer to the new file location. Which will also cause the git update mechanism to not work. We could ofcourse move this to a config file, but then you would still be required to edit that config file.

Mte90, 2017-10-18:
The important that is possible to change that logo, a config file is not a problem for me :-)