BillyBlaze/OctoPrint-TouchUI #132: Wrong 'main color' picked up when running multiple OctoPrint instances on he same server.

joris57, 2017-07-02:
I am running multiple instances of OctoPrint on different ports of the same (Ubuntu) server. To differentiate between them, I set a different 'appearance' color in OctoPrint.

In TouchUI, I set a matching 'main color'.

However, this color is not correctly displayed in Chrome (usually the same color from one of the three instances is displayed for all 3 instances). The 'regular' interface picks up the 'appearance' color correctly.

BillyBlaze, 2017-07-08:
Thanks for reporting, sorry but this is simply not possible since the code was not made with the intention of having multiple instances on one OctoPrint.

To support this; It would require a major refactoring of the code on the backend-end (Python) and the mechanism on the front-end (Javascript) that compiles the LESS files into a CSS file.

However, since I don't have much time on my hands at the moment, I can't implement this soon. I also needs some "standard documentation" about setting up multiple instances so I can rely on a consistent mechanisme for storing the and serving the right compiled CSS file.

joris57, 2017-07-08:
Thanks for considering this.

Instructions for setting up multiple instances can be found here : [https://github.com/foosel/OctoPrint/issues/113](url). This is the only 'standard' I have found.

Note that TouchUI seems in general to work well on multiple instances, with exception of this color issue.

gdachs, 2017-12-06:
You could use a docker container with octoprint inside. You can start multiple containers with just different ports. They will have their own file systems, so different colors would be no problem.

BillyBlaze, 2017-12-12:
Thanks for posting @gdachs, this is a very good alternative, therefor I am closing this issue.

Once OctoPrint officially support multiple instances I am willing to reconsider any changes needed with customization.

joris57, 2017-12-13:
I understand that the developer does not want to put in significant effort for the implementation of what certainly is merely a cosmetic issue, but calling having to learn about 'docker' a 'very good alternative' is cutting it a bit short.

BillyBlaze, 2017-12-13:
yeah, your right. After reconsidering it, I think I could easily fix this by appending OctoPrint's port number to the filename. You do run the same instance on two different ports right?

joris57, 2017-12-14:
I'm not sure if I interpret 'instance' in your question correctly, but I guess the answer is 'yes'.

I'm starting the 'octoprint' daemon multiple (3 at the moment) times with a different basedir, config file and port.

BillyBlaze, 2017-12-15:
Yeah, I meant OctoPrint instances. I could write the customization for each instance to a file like `touchui.custom.5000.css` instead of `touchui.custom.css`. That wouldn't require a lot of refactoring too. Most of the pain is going to be in migrating old settings and files into new ones that also include the port.

Anyhow; I'll add this in the next release.

BillyBlaze, 2018-01-15:
Sorry, was adding a new bot to github, you can ignore what he said ofcourse.

closed 2018-09-15 by BillyBlaze

BillyBlaze, 2018-09-15:
Thanks, this has been released in 0.3.12

commit 9b88868b (2018-09-09): :art: Support customization of multiple instances

Fixes #132