BillyBlaze/OctoPrint-TouchUI #134: Request/Question

CWRRT, 2017-07-07:
Is it possible to reorder tabs? I use an enclosure plugin and would like it to be on the main display without going through the hamburger menu. I was able to change this on regular web interface by changing config.yaml, but does not reflect on TouchUI.

Also, is there a way to change how the temp graph is displayed? Using the 7" raspberry touch screen and the graph looks rather smooshed vertically.

Thanks and great work!

BillyBlaze, 2017-07-08:
Hello,

Thanks for asking.

1) It is currently not possible to re-order the tabs, the order is hard coded in javascript. I will mark this as a future request, since it could be possible to honor the order of tabs from OctoPrint.

2) You could include your [own LESS file](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Customize:-Use-your-own-file) where you can overwrite the height and max-height of temperature graph. It is currently optimized for all screen devices, but I can imagine on some screen resolutions you want.

BillyBlaze, 2017-10-17:
It is now possible to re-order some of the tabs in the 0.3.9 release.

Thanks for reporting!

closed 2017-10-17 by BillyBlaze

tedder, 2018-07-08:
@BillyBlaze  this is marked as done and 'possible' but there's no commit, or instructions on **how** to do it. Can you give any info?

BillyBlaze, 2018-07-09:
Hi @tedder 

You can change the order of the tabs in OctoPrint through the config.yaml. An example of that is available [here](http://docs.octoprint.org/en/master/configuration/config_yaml.html#appearance).

TouchUI will honor that order.

CTCartesian3D, 2018-07-25:
How do you adjust the temperature-graph height... new to pi and linux and css is beyond me I think... 

Cannot work out how to do it or add a less file to override the default?

Apart from the graph everything else is looking great on the official pi display...

Would appreciate some help, even if its how to do and I'll happily have a go at it!

Thanks.