BillyBlaze/OctoPrint-TouchUI #135: Cannot print from Pi3 with TouchUI, and Controls are not displayed in Controls tab

hammiesam, 2017-07-12:
Hi Billy, 

Thank you so much for your work in this, it's amazing!

I'm running a Pi 3, adafruit 3.5" pitft, with the pitft image from Adafruit.

TouchUI boots up and everything, it's connected to my Prusa i3 MK2S, and gets the telemetry just fine.

However, even though I can see the files stored on my octoprint server, the PRINT icon is grayed out, and I cannot print from the TouchUI.  I can print from octoprint on my desktop browser and from my printoid app (android) just fine.

Am I supposed to be able to print from TouchUI?

Also, the controls are not displayed in my CONTROLS tab on TouchUI.  It doesn't seem to be loading.

Am I doing something wrong?

Sorry to bother you, I've tried to search on Google, but cannot find people with the same problems as I.

Thank you for your time!

Sam

BillyBlaze, 2017-07-12:
Hi Sam,

Your welcome and thank you.

First off; are you really, really sure you're logged in? it sounds like either you're not logged in, or that there is a bug during loading. If it's the latter we need to investigate some more.

hammiesam, 2017-07-13:
Ah yes, I was indeed logged out when the controls disappeared.  Now the problem is that the "remember me" function doesn't work, I've restarted the Pi and it's still the same.  The "remember me" check box does not have a check mark when I click it, the color of the check box will darken slightly, but if I restart the pi, it doesn't remember my login info.

Thanks for the awesome app!  Keep up the great work!

BillyBlaze, 2017-08-12:
Sorry for the long delay.

TouchUI does nothing with the login, this is provided by OctoPrint. Any issues with the 'remember me' token is done by OctoPrint.

You could enable autoLogin on localhost by changing your OctoPrint configuration. See more details [here](https://github.com/foosel/OctoPrint/wiki/Configuration#access-control)

closed 2017-08-12 by BillyBlaze