BillyBlaze/OctoPrint-TouchUI #137: TouchUI isn't auto booting /Xwrapper.config: No such file or directory

qpens, 2017-07-15:
Hi after step by step from your instruction from https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

raspi is not booting UI stuck on login

root@Axiom:/home/pi# apt-get install --no-install-recommends xinit xinput xserver-xorg xserver-xorg-video-fbdev x11-xserver-utils matchbox unclutter chromium-browser
Reading package lists... Done
Building dependency tree
Reading state information... Done
matchbox is already the newest version.
unclutter is already the newest version.
x11-xserver-utils is already the newest version.
xinit is already the newest version.
xinput is already the newest version.
chromium-browser is already the newest version.
xserver-xorg is already the newest version.
xserver-xorg-video-fbdev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@Axiom:/home/pi#

root@Axiom:/home/pi# sudo sed -i 's/allowed_users=console/allowed_users=anybody/' /etc/X11/Xwrapper.config
sed: can't read /etc/X11/Xwrapper.config: No such file or directory
root@Axiom:/home/pi#

How can i solve this problem?

BillyBlaze, 2017-07-16:
Hi qpens,

You can ignore the error of Xwrapper.config. Could you share the contents of /var/log/xorg.0.log

BillyBlaze, 2017-08-12:
Closing this issue due to inactivity.

closed 2017-08-12 by BillyBlaze