BillyBlaze/OctoPrint-TouchUI #138: Issue with Adafruit 7" HDMI Backpack and touchscreen

wookiemage, 2017-07-17:
Hello @BillyBlaze,
I am experiencing an inability to scroll with the current version of octopi (0.14) and the current TouchUI Plugin (0.3.8).
I am touching the screen, but it appears to be interpreting all touches as mouse clicks. So when I touch and drag it highlights portions of the screen instead of scrolling. Additionally, while I see a small blue scroll bar on the right hand side of the screen, I can't seem to grab it to move it.
I have run through calibration and can press other buttons accurately. 

This is what my current chromium.xinit reads:

```
#!/bin/sh

# DEFAULT VARIABLES
TOUCHUI_ARGS=""
TOUCHUI_DEBUG=""

# Uncomment the next variable to generate a full log under ~/.config/chromium/chrome_d$
# or to enable the remote deugger, forward port by haproxy or by tunnel
#TOUCHUI_DEBUG="--remote-debugging-port=9222 --enable-logging --v=100"

# Load octoprint settings for getting $PORT
[ -r /etc/default/octoprint ] && . /etc/default/octoprint

# Load touchui settings for getting $TOUCHUI_USER and $FORCE_TOUCH
[ -r /etc/default/touchui ] && . /etc/default/touchui

# Remove restore last session after crash in Chrome
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' /home/$TOUCHUI_USER/.config/c$
sed -i 's/"exit_type":"Crashed"/"exit_type":"None"/' /home/$TOUCHUI_USER/.config/chrom$
# And another one for not minified Preferences (Chrome legacy)
sed -i 's/"exited_cleanly": false/"exited_cleanly": true/' /home/$TOUCHUI_USER/.config$
sed -i 's/"exit_type": "Crashed"/"exit_type": "None"/' /home/$TOUCHUI_USER/.config/chr$

# Check if $FORCE_TOUCH is set to true
# This variable is set inside /etc/default/touchui
if [ "$FORCE_TOUCH" = true ]
then
        echo "TouchUI: touch events enabled (forced)"
        TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
else
        # Force Chrome to render with touch events
        # Bugfix #46
        TOUCHUI_DEVICE=$(xinput --list --long | grep XITouchClass)
        if [ -z "$TOUCHUI_DEVICE" ]
        then
                echo "TouchUI: touch events automatically detected"
        else
                echo "TouchUI: touch events enabled"
                TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
        fi
fi

CHROME_VERSION=$(dpkg -s chromium-browser | grep "Version: " | sed -e 's/Version: //g')
CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=10$
                --disable-java --disable-plugins --disable-extensions --disable-infoba$
                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT)'
                --start-maximized --window-position=0,0 
                $(dirname "$0")/load-screen/startup.html"

[ -r /home/$TOUCHUI_USER/TouchUI-autostart/calibration.sh ] && . /home/$TOUCHUI_USER/T$

# Start window manager, unclutter to remove cursor and start chromium as specific user$
matchbox-window-manager &
unclutter -idle 0.1 &
su $TOUCHUI_USER -c "chromium-browser $(echo $CHROME_ARGS|tr -d '\n')"
```

Here is the result of cat /proc/bus/input/devices
```
I: Bus=0003 Vendor=04d8 Product=0c02 Version=0111
N: Name="Microchip Technology Inc. AR1100 HID-MOUSE"
P: Phys=usb-3f980000.usb-1.3/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/0003:04D8:0C02.0001/input/input0
U: Uniq=
H: Handlers=mouse0 event0 js0 
B: PROP=0
B: EV=1b
B: KEY=70000 0 0 0 0 0 0 0 0
B: ABS=3
B: MSC=10

I: Bus=0003 Vendor=258a Product=0001 Version=0110
N: Name="SINO WEALTH USB KEYBOARD"
P: Phys=usb-3f980000.usb-1.4/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:258A:0001.0002/input/input1
U: Uniq=
H: Handlers=sysrq kbd leds event1 
B: PROP=0
B: EV=120013
B: KEY=10000 7 ff9f207a c14057ff febeffdf ffefffff ffffffff fffffffe
B: MSC=10
B: LED=7

I: Bus=0003 Vendor=258a Product=0001 Version=0110
N: Name="SINO WEALTH USB KEYBOARD"
P: Phys=usb-3f980000.usb-1.4/input1
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:258A:0001.0003/input/input2
U: Uniq=
H: Handlers=kbd event2 
B: PROP=0
B: EV=13
B: KEY=2a0 0 0 0 0 0 10000 2000000 3878 d801d001 1e0000 0 0 0
B: MSC=10

I: Bus=0003 Vendor=0000 Product=0538 Version=0111
N: Name=" USB OPTICAL MOUSE"
P: Phys=usb-3f980000.usb-1.5/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/0003:0000:0538.0004/input/input3
U: Uniq=
H: Handlers=mouse1 event3 
B: PROP=0
B: EV=17
B: KEY=70000 0 0 0 0 0 0 0 0
B: REL=103
B: MSC=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="ADS7846 Touchscreen"
P: Phys=spi0.1/input0
S: Sysfs=/devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input4
U: Uniq=
H: Handlers=mouse2 event4 
B: PROP=0
B: EV=b
B: KEY=400 0 0 0 0 0 0 0 0 0 0
B: ABS=1000003
```

I've tried adding the --touch-events=enabled flag to the list of chromium flags, but it doesn't seem to make any difference.

I have also attempted the suggestion by @zarderxio but when it didn't work, I reverted to stock settings. 
I am out of things to try.
Can you please help?

Thanks,

BillyBlaze, 2017-07-17:
Hi @wookiemage 

What happens if you specify ```--touch-events=disabled```?

The bluebar on the right indicates that the browser is listening for touch events, but it seems like your driver is emitting mouse events.

Do you still have small blue bar on the right after this?

wookiemage, 2017-07-17:
I made that change and it does remove the blue bar. 
Sounds like you're right about the driver emitting mouse events. 

Should I reenable the touch-events? And is there some way to change how the driver works? I know this is probably well beyond your support, but I appreciate all the assistance you can provide.

wookiemage, 2017-07-18:
@BillyBlaze I found a workaround that solved my problem. I enabled extensions and installed ScrollBars Anywhere (a chrome extension that emulates scrolling behavior seen on a touch device through a mouse controller).
I had to comment out a lot of flags in the chromium.xinit file in order to download and install the extension. The only lasting change I made to the chromium.xinit file I made is to comment out the --disable-extensions flag.

Now I can use the screen to control everything. Thanks again for your help!

closed 2017-08-12 by BillyBlaze