BillyBlaze/OctoPrint-TouchUI #141: A get info page

drandolph1976, 2017-07-30:
Since I've setup TouchUI with a touch screen I no longer can see the IP address or name of the server in the prompt. I would like a way to see the IP address from the touch screen without a keyboard.

bakerboy908, 2017-08-01:
Log in to your router, there should be a LAN page that will show you all IP's of all devices on your network.

BillyBlaze, 2017-08-12:
Hi,

Thanks for submitting the feature request.

I will not implement any additional functionality to this plugin beside anything related to it's core functionality. e.g. providing a mobile and touchscreen friendly interface.

However, since TouchUI only transforms the layout you can still install additional plugins on OctoPrint to show the IP address.

Please feel free to submit a feature request to support a specific plugin.

closed 2017-08-12 by BillyBlaze