BillyBlaze/OctoPrint-TouchUI #142: General question about color customization

devsfan1830, 2017-08-11:
I'm a bit lost as to where to change the variables to change UI colors. The animation in the main page shows a settings area that I don't seem to have. There is no section that shows color variables. Did this feature change. Is there a file I can edit?

BillyBlaze, 2017-08-12:
Hi, 

Nothing has changed, you should be able to view the TouchUI settings under ```Settings > TouchUI```

If the setting section is not there, then please share your octoprint log file located under ```~/.octoprint/logs/```

devsfan1830, 2017-08-12:
My apologies. There was no obvious indication that the settings page had different sections browsable by tapping the text at the top. I found it and it works as advertised. Thanks!

closed 2017-08-12 by devsfan1830

BillyBlaze, 2017-08-12:
No apologies needed, I can imagine that the feedback isn't clear enough. I will improve this in the future release so that it's more clear that you can click it.

& Your welcome!