BillyBlaze/OctoPrint-TouchUI #143: Capacitive Touch Screen Improvement Request / Help

wookiemage, 2017-08-23:
We are currently running Octopi 0.14 and TouchUI 0.38 on a fleet of 10 Lulzbot minis, controlled locally through the [image] which treats touch input as mouse clicks. 
In order to get scrolling working through the touch display I had to install a chromium plugin [image]
That's all fine and good, but this introduces a little difficulty when actually using the screen with your finger. Namely, when you try to press an icon (say the print icon) it can be difficult to get the screen to register that as a click instead of a scroll. 
I was hoping you might have some ideas about what to customize or change to make this easier to use. Here are a few things I am considering:
* Could I make that icon larger / make the touch target larger? (This would help, but we would probably still encounter strange scrolling due to the combination of the plugin and TouchUI) 
* Could I change the default behavior for each listed item so that tapping anywhere on that line starts a print? [image]
* Some other way to make this experience better that I haven't thought of?

I know that you have better touch screen support listed as a 1.0 goal, so I understand if there isn't anything I can change at the moment.

Thanks for your help

BillyBlaze, 2017-08-24:
Hi @wookiemage,

1. In the upcoming release I will introduce a new variable that will influence all the font sizes including icon sizes. If this is not enough, you can also include your own LESS file and overwrite the sizes with CSS yourself. [Some more info about this here](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Customize:-Use-your-own-file)
2. Within TouchUI, no, thats not possible, however you can easily setup a plugin yourself and include a small javascript file that helps, You could ofcourse also try to install an extension like GreaseMonkey.
3. To be honest, if I knew one then I would try to implement it. However in your situation I think your biggest experience gain would be to get your devices to emit touch events. Sadly that is something out of my reach to help you with.

wookiemage, 2017-08-25:
@BillyBlaze 
Thanks for responding. I'll look into either including my own LESS file or potentially setting up a plug in to alter the behavior of a row. 
Thanks for writing back and for making such a great plugin!

BillyBlaze, 2017-08-25:
Your welcome and thank you too 👍 

If you need any help with JS or LESS just throw a message in this issue.

closed 2017-08-25 by BillyBlaze