BillyBlaze/OctoPrint-TouchUI #144: Stuck at TouchUI logo

number12, 2017-08-24:
I followed all the steps and while doing so I couldnt get step 6 to work. So I skipped it for then to get the rest done. After that did rboot and not stuck at the touchui logo. Tried to do step 6 again but it says it cannot find that file/directory. So I check the dir and I dont see it either and I didnt see a step to create the file.

> pi@octopi:~ $ sudo sed -i 's/allowed_users=console/allowed_users=anybody/' /etc/X11/Xwrapper.config
`sed: can't read /etc/X11/Xwrapper.config: No such file or directory`

BillyBlaze, 2017-08-24:
Hi @number12 

You can ignore this step. Can you describe your situation a bit more detailed? Such as the status message beneath the logo?

Ipoc78, 2017-08-25:
HI BillyBlaze. I am in dire need of your help, I was just about to open a new issue but found this one thats kinda in the same ball park as mine. 

First I have done your "Setup: Boot to Browser (OctoPi or Jessie Light)" and everything worked fine the first time around anyway.  

the only other thing i did was add lcd_rotate=2 in /boot/config.txt 
every thing work just right even the touch (not backwards)

I had added a plugin to octprint and refreshed the screen and then i was switching from the chrome lay out back to the Touch layout  and thats when things went crazy... 
after a reboot, the touchui comes up with "startup failed, tap to retry" under that "uncaught refernceError: $items is not defind"

SO.. i did a full reinstall thinking there is a bug with a plugin, this time only plugin i added was for Touchui,
this  time it all came up good but.. now my pointer runs opposite of my finger >.> 
as long as i dont do a "lcd_rotate=2" it was ok

so i was trying to do your setup to "Calibration rotate touchscreen" and well things went even more out of hand

SO i did yet another full reinstall but this time i didnt mess with the "lcd_rotate=2" 
it all booted up fine but when i did a refresh BAMN!! right back to this
    the touchui comes up with
   "startup failed, tap to retry" 
   "uncaught refernceError: $items is not defind"

Im so lost on whats going on and why. im not sure why it worked just right the very first time and have so many problems with it now,  HELP!!!!

BillyBlaze, 2017-08-25:
Hi @Ipoc78 

Thanks for reporting. I think I messed up and I can see why it could throw an error on $items.
I will investigate this some more this weekend.

Ipoc78, 2017-08-25:
Thank you so much! 

For reference: I'm running the latest version of octpi 0.14 on a raspberry pi3 and using the official pi 7" touch scree
Sidenote i did not have any issues on octpi ver 0.13  (i think that was the last ver i was running, I had a mishap with an SD card and had to start all over :(  )

number12, 2017-08-26:
Well It shows a Blue logo that says TouchUI with small writing underneath, too small for me to read, then after a while it changes to red. So that is what Ive getting atm. Not sure if that helps

Ipoc78, 2017-08-26:
[image]

Here's a snap shot of mine, look like that ? 
The red comes after the blue like yours.

number12, 2017-08-28:
Ok im getting same thing as Ipoc78. Im on latest kernel and updates for pi. I followed all the steps the only thing is that even though im using this monitor https://www.amazon.com/Raspberry-Display-kuman-480x320-Interface/dp/B01CNJVG8K its registering as an adafruit lcd as dev0 not dev1. Not sure if that is what causing that issue.

Ipoc78, 2017-09-03:
Is there going to be a fix for this soon? or do I need to move on to something else ?
I've tried over and over to do a fresh install and it always ends up stuck on the picture above..
is there anything else i can try?

BillyBlaze, 2017-09-03:
I have tried to reproduce the problem, but I just cannot get the same error. I did identify a possible culprit, but I am unsure if this will resolve your issue. Never the less you could try to see if it works for you;

1. Uninstall TouchUI from OctoPrint
2. Install TouchUI by link with ```https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip```
3. Reboot

Ipoc78, 2017-09-05:
Well sir.... Im at a loss for words, after about a week its been setting there I plunged it in after your reply and it works!! I have no idea how or why... but  I did go ahead and uninstall and install from the link you gave, so for now problems with boot up, Im still having an issues with rotating the screen and the pointer working with my finger (y/x is backwards) the problem i was having was i could not get it to write a new file using  
sudo nano /etc/X11/xorg.conf.d/99-calibration.conf
after entering Section until Endsection from the calibration i did 
Save & Exit (CTR-O, ENTER, CTR-X)
when i did a CTR-O and then enter, it gave me an error and was not able to write (after many attempts)
at this point Ive lost most of my hair...  I was able to relocate how i was mounting the screen and just went with that.   lol 

Thanks for your help. as of now the issue with boot up/screen stuck  is gonna

Ipoc78, 2017-09-05:
Well I spoke too soon I came home from work plugged in my printer and booted up to this
[image]

winchymatt, 2017-09-13:
Hi BillyBlaze.

Thank you, this worked for me! :)

Matt.

commit e48a6a6b (2017-09-03): :bug: Declare local variable

Do not write $items to the window object.
Might fix #144 and #145
closed 2017-10-17 by BillyBlaze